LibreCAD/LibreCAD #1344: Copy paste block from clipboard adds new block names to block list

allanwmacdonald, 2021-05-02:
Expected Behavour: Pasting blocks from the clipboard creates new instance of the original block name and the block list does not change.

Observed Behavour: Pasting blocks from the clipboard creates a new block name in the block list and the block list grows with every pasted block entity.  If multiple objects are pasted, each new block produces a new name in the block list and the list can grow very large.

To recreate the issue:

1. Create a drawing with just one entity: a single block composed of some primitive entites like arcs and lines, such as a door symbol.

2. In the block list, confirm there is just one block listed.

3. Using the block list, it is possible to insert as many instances of that block as you want and the block list does not change.

4. Using the Move/Copy command and perform a copy (i.e. move, retain original), again, it is possible to make as many new instances of the original block as you want and the block list does not  change.

5. BUT, by selecting an instance of the original block, and performing a ctrl-c operation (i.e. copy to clipboard) and then pasting into the same drawing, a NEW BLOCK is created in the block list.

Environment:
LibreCad Version: 2.2.0-rc2-15-g76828565
Compiler: GNU GCC 9.3.0
Compiled on: May  1 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.2 LTS

See this forum topic: https://forum.librecad.org/Librecad-keeps-adding-blocks-to-block-list-td5715315.html#a5720508

melwyncarlo, 2021-11-19:
From what I've read, that seems to be the intended function: to copy blocks, and not mere drawings within them.

It would be better if the _Blocklist_ had its own Duplicate option (button), and if the ordinary Copy/Paste operation would be redirected to the Move/Copy operation instead.

allanwmacdonald, 2021-11-19:
@melwyncarlo:

Thanks for your reply.  Please provide a reference to the information or documentation that contains the meaning you have come to understand.

Intuitively, the elements in the drawing area are _instances_ of a particular entity, such as a line, shape or block (a block being a compound entity).  Copying lines or rectangles do not create new definitions of entites (i.e. line, line-1, line-2, etc.) so why should copying a placed block instance do so?  Why would the rules change based on what kind of object is being copied?

When a block is inserted in a drawing via the block commands, _instances_ of a block are created.  Why does the copy/paste operation do something different?  What's being copied and pasted?  Is it the instance or the definition?

Yes, I agree that there ought to be a duplicate block feature.  However, in my opinion, copying an _instance_ of a block should not create a new _definition_ of a block.  If this was the intended behaviour, the documentation ought to be more specific as to what you are copying using the copy/paste functionality.

I guess this all comes down to the semantics of the word "block".  Does this word refer to the definition of a block or its instance?  Is an entity an instance of something or its definition?

As far as whether or not this is an important issue, think of the usability.  Imagine someone designing a structure that has a number of identical objects, such as windows, for example.  Say the user placed a group of such object into a pattern, copied the entire pattern and pasted the same pattern elsewhere on the drawing.  Now you have a bunch of new block definitions you didn't want, along with an unnecessary increase file size, complexity, etc.

So, if this was the "intended behaviour" as you say, perhaps a different choice could have been made in the design of this tool.

GSStnb, 2021-11-20:
@melwyncarlo

"I guess this all comes down to the semantics of the word "block". Does this word refer to the definition of a block or its instance? Is an entity an instance of something or its definition?"  <- It does make sorting through this a little awkward, however...

I agree with  @allanwmacdonald.  Copying an instance of a block from within a drawing that appears in the **Block List** _should NOT_ result in another block being added to the list, but should be another instance of the same block.  As it currently is, copying a block breaks the behaviour of a block (e.g. editing one block instance should change them all).  Blocks added from the **Block Library** _should_  (and do) result in a _new_ block being added to the Block List.

Again, it might be semantics, but perhaps "cloning" a block in the Block List would be a better option when creating a new block from an existing one.  Once created, it can become its own thing.

melwyncarlo, 2021-11-28:
Video link : [https://youtu.be/Aarju_tMJIQ](https://youtu.be/Aarju_tMJIQ)

JeremyRuhland, 2022-07-31:
Tested with [5a07ab4](https://github.com/LibreCAD/LibreCAD/commit/5a07ab418ab23f2f2553d267eeb15a09c1220ec0) and working great, thank you! This weird behavior had been driving me crazy for years.

lordofbikes, 2022-08-01:
@JeremyRuhland what exactly have you tested?
You referenced 5a07ab4, which is in fact the latest commit on `master`.
But the fix for this issue, #1437 is not merged yet.

JeremyRuhland, 2022-08-01:
I merged #1437 into a local copy of the repo and compiled. (Maybe I should have posted this to the PR comment thread? I'm always torn between the issue tickets, feature requests, PR comments and the forum threads.)

Copy/paste of blocks within a document and between documents is working as described in the PR and matches the behavior of all the other 2D CAD programs I've used.

lordofbikes, 2022-08-01:
No problem.
In a PR comment I'd expected that you have merged and build, but here I was not sure and asked for clarification.
Many thanks for testing and reporting! :+1: