LibreCAD/LibreCAD #1347: Large DWG reading errors

arturred, 2021-05-18:
Hi
I get crashes and errors when opening large dwg files in Windows. Here is a link to sample data : https://easyupload.io/3yjmfh
I also found a ton of files that fail in this same manner : https://www.3drotterdam.nl/downloads/#/
I either get duplicated ids of a page map section or invalid addresses outside the buffer range. In other files, the page map seems to be correct but reading section info fails. The problem is not only overflow values but also the decompressed buffer that may contain gaps (negative page id?).

Steps : 
download https://www.3drotterdam.nl/downloads/global/download//DWG/Rotterdam_Centrum.zip
unzip Rotterdam_Centrum.zip
read Rotterdam_Centrum/Bomen/Cool.dwg
or other bigger files.

download https://easyupload.io/3yjmfh
unzip dwg
read error?.dwg files
I added extra ?2000_ok dwg files that open correctly as reference.

Is libdxfrw still developed and these errors can be fixed? I also tested libredwg but this library also fails to open such files.

fa201, 2021-05-18:
I tried the with a fresh build.
Version: 2.2.0-rc2-15-g76828565
Compiler: GNU GCC 7.5.0
Compiled on: Apr 17 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

The error message is 
[image]

lordofbikes, 2021-05-19:
This is still tracked in #988, even when LibreCAD don't crash with this sample.
There are several issues with samples cross linked there, including this now.

I'll take your samples in account for further tests, many thanks for reporting.

Yes, the goal is to make libdxfrw more stable and able to read more DWG files without errors.
The problem is, that DWG is still proprietary and public references are from reverse engineering and therefore incomplete inherently.

closed 2021-05-19 by lordofbikes