LibreCAD/LibreCAD #1348: Suggestion: Possible to have properties menu always showing on toolbar(example on the right side toolbar)

Decee1, 2021-05-20:
I am in the process of switching from Draftsight to Librecad and what i felt horrifying is i am missing my XY coordinate menu which i use a lot. 
So every time now i have to change a line i have to right click and click properties instead of just clicking on the line and it will show the properties.
So i would like to suggest a feature which allows the properties menu to have a fixed position on the toolbar. 
Specially because the properties menu is a popup box and everyone hate popup boxes! :P

lordofbikes, 2021-05-20:
duplicate of #622
I can't promise that this will be realized in V2 branch.
LibreCAD_3 will have it, but we still have no schedule for a pre-release yet.

closed 2021-05-20 by lordofbikes

Decee1, 2021-05-20:
Great to hear! Can't wait for that to be added

gerardbm, 2021-11-22:
+1 for this! Should this be tagged with the `v3` tag, @lordofbikes?