LibreCAD/LibreCAD #1349: Add new glyphs to the vectorfont from Solvespace

phkahler, 2021-05-20:
Solvespace uses the vector font from LibreCAD and has had some character added since then:
https://github.com/solvespace/solvespace/issues/1042#issuecomment-844820653

Some Ukrainian characters as well as the Depth, Counterbore, and Countersink commonly used to annotate holes.

I can make PR to merge these into the current LibreCAD font file, or we could go the other way and merge LibreCAD changes into solvespace and give them back. It seems like a good idea to share these updates between projects. If we get back in sync I'll try to remember to submit changes to both if/when they happen again.

Symbian9, 2021-05-20:
JFTR, I'm an author of those *"Some Ukrainian characters"* (i.e.`Ґ`, `ґ`, `І`, `і`, `Ї`, `і`, `Є`, `є`), which are already merged in LibreCAD (and then was merged to SolveSpace too!):

- https://github.com/LibreCAD/LibreCAD/issues/690

But @Evil-Spirit later found small issue with some of those characters (e.g. `є` and `Ї`):

- https://github.com/LibreCAD/LibreCAD/commit/ca6f39feb916390596c7ed3a0cfd88e935cd4e74#commitcomment-17007934
- https://github.com/LibreCAD/LibreCAD/commit/ca6f39feb916390596c7ed3a0cfd88e935cd4e74#commitcomment-17007980

And there was an extra patch to `є` charater added in SolveSpace added:

- https://github.com/solvespace/solvespace/commit/9c08398f51d5040245001c53ccd8388107221216#diff-e99cc430914390ed01b41b9eeefd419ec4c099e6b15d4fd5b29612e21cb8cfa0

So, ***extra check both*** versions of `unicode.lff` (SolveSpace's `master` & LibreCAD's `master`) before merge each to another or vice versa!

app4soft, 2024-08-31:
@dxli This issue could be closed as solved.