LibreCAD/LibreCAD #1352: (Suggestion) Insert Polyline Break Line

ChickenHacker12, 2021-05-26:
I was wondering if a poly line break line feature could be added (like the one in QCAD). It allows a break line to be created between two points. This would be great for engineers and architects with large drawings.

Questions2000, 2024-09-29:
@sand1024 
This is on list of issues to close, had this been added?
It does not mentioned that it has been "fixed" "added"

sand1024, 2024-09-29:
well, this feature does exist for normal lines - normal break, Break/Divide between intersection points, Line Gap etc. 
For polyline - it's possible to break using Divide command.

Honestly, actually I'm not sure I not fully understand what exactly was requested by original issue. 

For sure, potentially, polyline may be considered for gap in the way similar to ordinary line. However, I suspect that it might be overkill for majority of cases... 

Anyway, if special handling of polylines is required - it's better to close this issue and create the new and more focused one.

closed 2026-09-26 by sand1024