LibreCAD/LibreCAD #1357: "Multiple Copies" in "Move/Copy" should be considered as one action and be cancled by one ctrl-z

WhymustIhaveaname, 2021-07-02:
### Expected behavior

Multiple Copy anything, for example, 40 copies. Then using ctrl-z to cancel this action. Intuitively, this is one action so one ctrl-z is enough. Actually, I have to use 40 ctrl-z's to cancel it. I can understand reusing the function of "Keep Original" mode in "Move/Copy" many times to realize "Multiple Copy". However, not modifying histories is irresponsible.

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

```
Fedora 33 (Workstation Edition)
```

```
Version: 2.2.0-alpha
Compiler: GNU GCC 10.2.1
Compiled on: Jul 28 2020
Qt Version: 5.15.2
Boost Version: 1.73.0
System: Fedora 33 (Workstation Edition)
```

lordofbikes, 2021-07-02:
Fixed in latest code

closed 2021-07-02 by lordofbikes

WhymustIhaveaname, 2021-07-02:
Oh, thanks.