LibreCAD/LibreCAD #1358: Cannot find auto-backup file

WhymustIhaveaname, 2021-07-02:
### Expected behavior, Observed behavior

I have turned on "Auto backup" in "Application preference". However, after the crash of this app, there is no hint of where can I find it!

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

```
Fedora 33 (Workstation Edition)
```

```
Version: 2.2.0-alpha
Compiler: GNU GCC 10.2.1
Compiled on: Jul 28 2020
Qt Version: 5.15.2
Boost Version: 1.73.0
System: Fedora 33 (Workstation Edition)
```

lordofbikes, 2021-07-02:
https://librecad.readthedocs.io/en/latest/getstart/configure.html?highlight=backup#program-defaults

closed 2021-07-02 by lordofbikes

WhymustIhaveaname, 2021-07-02:
The document reads
```
Auto backup: When checked, a backup will be created when closing the file. Backup files are saved to the same directory as the drawing file with a tilde (~) appended to the file name.
```
Unfortunately, __this is not my case__. I have not saved it although I have been editing it for a long while. And because I have been editing it for a long time, not able to recover it after crush is depressing.

lordofbikes, 2021-07-02:
Then you possibly were hit by #1086 and may find the auto save file in /tmp.
Sadly your steps to reproduce didn't mention, that you worked on a fresh, unnamed file.
Generally it's bad practice to work long time on a unnamed file. This concerns not only CAD, it is recommend to all kind of documents to name it as soon as possible. Especially when one put a bunch of effort into it.

WhymustIhaveaname, 2021-07-03:
Yes, I know that I should not work for too long without saving. It was my fault yesterday.

I had the same idea at the first sight that the file may in `\tmp`. So I `ls`ed `\tmp` but found no suspect.

Another motivation of this issue, after confirming sometimes LibreCAD saves files in `/tmp`, is to suggest not using this directory. Because more and more Linux distributes mount `/tmp` on RAM, this is no longer reliable. For example, my Fedora does so, and Ubuntu20 can mount `/tmp` on RAM with one simple command.

Hope LIbreCAD be better and better!