LibreCAD/LibreCAD #1366: Clicking "hatch" with selected rectangle causes segmentation fault

blaa, 2021-07-17:
### Expected behavior
Hatch should probably allow to fill rectangle with a color. I'm learning - unsure

### Observed behavior
I get segmentation fault.

```
Thread 1 "librecad" received signal SIGSEGV, Segmentation fault.
0x0000555555808083 in QG_DlgHatch::updatePreview (this=0x7fffffffcdc0) at ui/forms/qg_dlghatch.cpp:158
158		if (pattern->countDeep()==0)
(gdb) p pattern
$1 = (RS_Pattern *) 0x0
(gdb) bt
#0  0x0000555555808083 in QG_DlgHatch::updatePreview (this=0x7fffffffcdc0) at ui/forms/qg_dlghatch.cpp:158
#1  0x00007ffff6efb5e0 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff7a73ea2 in QAbstractButton::toggled(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff7a75019 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff7a757d0 in QAbstractButton::setChecked(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x0000555555807a20 in QG_DlgHatch::setHatch (this=this@entry=0x7fffffffcdc0, h=..., isNew=isNew@entry=true) at ui/forms/qg_dlghatch.cpp:99
#6  0x00005555557dfb58 in QG_DialogFactory::requestHatchDialog (this=0x7fffffffcdc0, hatch=0x7fffffffced0) at ui/qg_dialogfactory.cpp:1913
#7  0x000055555578e4d5 in RS_ActionDrawHatch::init (this=0x555560564520, status=<optimized out>) at actions/rs_actiondrawhatch.cpp:55
#8  0x0000555555719942 in RS_EventHandler::setCurrentAction (this=0x555555eb5110, action=<optimized out>, action@entry=0x555560564520) at lib/gui/rs_eventhandler.cpp:440
#9  0x000055555571a653 in RS_GraphicView::setCurrentAction (this=0x555555f8ef70, action=action@entry=0x555560564520) at lib/gui/rs_graphicview.cpp:256
#10 0x00005555557d02dc in QG_ActionHandler::setCurrentAction (this=0x555555e65700, id=<optimized out>) at ui/qg_actionhandler.cpp:947
#11 0x00007ffff6efb5e0 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff7980b62 in QAction::triggered(bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff79833a1 in QAction::activate(QAction::ActionEvent) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff7a761b2 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff7a762d1 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff7b659ca in QToolButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff79c6fae in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff798715f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff798e04b in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff6ec4fca in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff798d076 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff79df011 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff79e233e in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff798715f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6ec4fca in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff729f583 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#27 0x00007ffff727564c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007ffff382e7da in ?? () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#29 0x00007ffff5dd8e6b in g_main_context_dispatch () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff5dd9118 in ?? () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff5dd91cf in g_main_context_iteration () from /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff6f1c51f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00007ffff6ec398b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff6ecbc00 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x000055555563c9de in main (argc=<optimized out>, argv=0x7fffffffdef0) at main/main.cpp:363
(gdb) 

```
### Steps to reproduce or sample file
1. Create rectangle
2. Click hatch.
3. Segfault
4. Profit.

Tested on empty file as well.

### Operating System and LibreCAD version info
commit e02f3580311d50b617427652ff77988f6bd62d9d 
I've compiled it from github and when it happened pulled and rebuild this commit.

Debian bullseye, x86_64

blaa, 2021-07-17:
```
-       if (pattern->countDeep()==0)
+       if (!pattern || pattern->countDeep()==0)
                return;
```
Doesn't die, but I can select only one pattern and it doesn't seem to work (?) This might be my indolence.

pattern && ... version seems to generate correct preview. I still don't have patterns, but solid colors work.

blaa, 2021-07-17:
With configured Hatch path in settings to `/home/bla/_local/LibreCAD/unix/resources/patterns/` it works ok and maaaybee there wouldn't be segfault. Still maybe it's ok to secure it?

fa201, 2021-07-17:
If you want fill a pattern with color, the feature exists: click on **Solid Fill**
[image]

blaa, 2021-07-17:
Yeah, but before I fixed the code, the whole dialog didn't show up. It's working with the fix. If you want I can do Merge Request with it.

fa201, 2021-07-18:
@LibreCAD/authors can anyone check on debian the above issue ? I have never had this issue on Ubuntu.

lordofbikes, 2021-07-19:
@fa201, I have to check this in code base.
When I understood correct, @blaa built from source and so it may be an issue of improper installation/packaging.
Using distributes packages or running a self built version from librecad/unix folder shouldn't have this issue, because everything is in place and found by the application.
Anyhow, when a missing folder leads to a seg fault, we should fix this.
Many thanks to @blaa for figuring this out.