LibreCAD/LibreCAD #1368: Bug: Mouse icon is really big in v2.2.0-rc2

MarcelloTheArcane, 2021-08-02:
### Expected behavior

I'm pretty sure the mouse icon was much smaller before - I updated LibreCAD to get the print tiling feature.

In this PDF instruction manual you can see a much smaller mouse icon:
[image]

### Observed behavior

It's huge!  So much so that I think when it disappears for some actions, the rest of the application jumps around.

[image]

### Steps to reproduce or sample file

N/A - just open LibreCAD.  It's an application issue rather than specific file issue.

### Operating System and LibreCAD version info

Version: 2.2.0-rc2
Compiler: GNU GCC 7.3.0
Compiled on: Dec 31 2020
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

fa201, 2021-08-02:
Possibly linked to #1303, #1026, #1010.

MarcelloTheArcane, 2021-08-03:
Thanks @fa201 - sorry I didn't see those.

Shall I close this issue, as it's being tracked in others?

fa201, 2021-08-03:
I believe not as some report were supposed to be OS specific (Ubuntu). Anyway it is not link to the chain of issues related to this subject so it should be easy to close all of them when it is solved.

melwyncarlo, 2021-11-16:
The icon's size is implemented as is. The big mouse icon (located at `librecad/res/icons`) size was **80x80** pixels. The small mouse icon (located at `librecad/res/ui`) size is **24x24** pixels. I think _this_ icon was used before. To implement the right icon, just ensure that the icon size is about **25x25** pixels.

lordofbikes, 2022-06-04:
fixed with eb0223968761df405c16a4885fff36c2fd05e9d1

closed 2022-06-04 by lordofbikes