LibreCAD/LibreCAD #1370: Missing Sections in Application Preferences

Jack-Indaboks, 2021-08-02:
I have not been able to activate the command line without manually clicking on it first. I tried Space Bar, ctrl+m, etc, but nothing worked. I finally found what appears to be the answer in another issue:

```
Note also that you need to enable the SPACE BAR validation in the **Application Preferences** menu.

![image](https://user-images.githubusercontent.com/49532187/106368566-c348f400-634a-11eb-876c-7b53687ba50d.png)

_Originally posted by @fa201 in https://github.com/LibreCAD/LibreCAD/issues/1279#issuecomment-770284053_
```

But my application preferences menu lacks the entire "Keyboard Settings" and "CAD Preferences" sections. Are those sections supposed to be there or are they part of an add-on or something?

OS: Arch Linux
LibreCAD Version: 2.1.3
Compiler: GNU GCC 11.1.0
Compiled on: Jun 10 2021
QT Version: 5.15.2
Boost Version: 1.76.2

lordofbikes, 2021-08-03:
The referenced image is from a newer version, it shows the dialog in latest code.

Also the mentioned option to enable SPACE BAR is not about activating the command line, this should work always.
This option is to use SPACE like ENTER and evaluates commands when a blank is added to the command line.
But this applies also to latest code too, not implemented in 2.1.3

Iirc there were some issues fixed with command line since 2.1.3, but I don't remember if there was any issue with activation.

fa201, 2021-08-03:
@Jack-Indaboks CTRL+M should work on 2.1.3. At least it works on my build:
Version: 2.1.3
Compiler: GNU GCC 7.4.0
Compiled on: May 19 2019
Qt Version: 5.12.8
Boost Version: 1.65.1
System: Ubuntu 20.04.2 LTS

sand1024, 2026-09-26:
issue for too old version

closed 2026-09-26 by sand1024