LibreCAD/LibreCAD #1371: Import error: Error reading DXF/DWG sections

dpocock, 2021-08-03:
### Expected behavior

Open the DXF file or show a more detailed reason why it fails to open.

### Observed behavior

There are no details about the error

### Steps to reproduce or sample file

./unix/librecad my-file.dxf

### Operating System and LibreCAD version info

Debian 10

LibreCAD built from Git commit e02f3580311d50b617427652ff77988f6bd62d9d

lordofbikes, 2021-08-03:
There are no more details for this error.
It means, that an unexpected code was read.
This usually only happens with incompatible files created by other software.
Or you possibly found a bug.

Anyhow we need the sample file, which triggered this message. 
Or information how this DXF file was created.

dpocock, 2021-08-03:
```
$ ./unix/librecad /tmp/my-file.dxf 
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
QG_LayerWidget::update: nullptr lastLayer
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
RS_FilterDXF::addLayer: layer 0 have extended data
RS_FilterDXF::addLayer: layer _ex first (3D) have extended data
RS_FilterDXF::addLayer: layer _ex roof (3D) have extended data
RS_FilterDXF::addLayer: layer _ex ground (3D) have extended data
RS_FilterDXF::addLayer: layer _ex eles (3D) have extended data
RS_FilterDXF::addLayer: layer _ex OS (2D) have extended data
RS_FilterDXF::addLayer: layer _pr first (2D) have extended data
RS_FilterDXF::addLayer: layer _pr roof (2D) have extended data
RS_FilterDXF::addLayer: layer _pr ground (2D) have extended data
RS_FilterDXF::addLayer: layer _pr eles (3D) have extended data
RS_FilterDXF::addLayer: layer _pr OS (2D) have extended data
RS_FilterDXF::addLayer: layer None have extended data
RS_FilterDXF::addLayer: layer MacOS have extended data
RS_FilterDXF::addLayer: layer DETAIL have extended data
RS_FilterDXF::numberToColor: Invalid color number given.
Cannot open DXF file '/tmp/my-file.dxf'.
QG_LayerWidget::update: nullptr layerList
```

dpocock, 2021-08-03:
Is it possible for LibreCAD to log the unexpected code or any other details that may help to focus on the source of the problem?

dpocock, 2021-08-03:
Is this enough detail to confirm the way the file was created or is there a more precise technique I can use to extract details from a DXF?  It was created by somebody else and sent to me by email.

```
$ strings my-file.dxf  | grep -i autocad
AutoCAD Binary DXF
WipeOut|AutoCAD Express Tool|expresstools@autodesk.com
WipeOut|AutoCAD Express Tool|expresstools@autodesk.com
```

lordofbikes, 2021-08-03:
The only way to get more output is to set debug mode in `libdxfrw`
modify this line:
https://github.com/LibreCAD/LibreCAD/blob/e02f3580311d50b617427652ff77988f6bd62d9d/libraries/libdxfrw/src/libdxfrw.cpp#L37
to:
```c++
        DRW_DBGSL(DRW_dbg::DEBUG);
```
and then rebuild `librecad`

The command line will then be flooded with `libdxfrw` output, may be too much, but the only chance.

The fails can be so different, that it doesn't make sense to overpower the regular CAD user with a bunch of cryptic data.
Also the regular CAD user will usually not be able to fix any errors in DXF files.

Either it was an incompatible creator software or missing features in the reader when loading fails.
In both cases the best option is to give the DXF file in question to the devs of DXF reader or writer.

Beside that, I've never seen a binary DXF since I put my hands on LibreCAD. So this possibly could be an undetected issue in `libdxfrw` because binary DXF is really rarely used, at least in free software.
But from the `addLayer` output above my guess is some unsupported 3D or extended data which are not handled in `libdxfrw`.

dpocock, 2021-08-03:
Thanks for this feedback.

I've enabled the debug logging and sent you the output by email

If there is something relevant in the log feel free to share the necessary lines here in the issue

lordofbikes, 2021-08-04:
Many thanks for the debug log, but without the DXF file there is not much I can do. Feel free to send it like the log.

My first step, before fiddling in `libdxfrw` code, would be to open the file with [TrueView](https://www.autodesk.eu/products/dwg/viewers) to see if it generally opens and to have a reference about the content.
If this already fails, it doesn't make much sense to debug in LibreCAD code.

But probably it would load fine in TrueView, then we must find the cause in `libdxfrw`.
The issue is triggered in `dxfRW::processEntities`, in *BLOCKS* section of the file.
As it is just the 3rd call I would set a break point there and see what is going on in the 3rd call with the value of code 310.

The log output looks good until the content becomes binary garbage at the end.
The file version (AutoCAD 2010/2011/2012) should basically be supported by `libdxfrw`, but as already said, binary DXF is so rarely that it could be an undetected issue.
The job is now to find the location in the file and use the DXF 2000 reference to evaluate the expected behavior and see what's going wrong in `libdxfrw`.

dpocock, 2021-08-05:
Thanks for this feedback

I've been trying to work out the history of this file

Somebody sent the file as a DWG file in Nov 2018.  I have the DWG file as well and it doesn't open in LibreCAD.

It looks like somebody else opened the DWG file and saved it as a DXF file.  This opening and saving as DXF might have been done on AutoCAD 2017.

I tried opening the DWG file in LibreCAD and that also fails.

I could do either of the following:
- go back to the person who originally sent the DWG and ask for them to save it as DXF in whatever tool they use,
- try to open it in AutoCAD 2017 again and save it as DXF again - are there any options or settings in AutoCAD that should be used when trying to do this again?

Also, if I send you the DXF then I probably need to send you the DWG as well.

lordofbikes, 2021-08-05:
The DWG support in `libdxfrw` is much less complete than DXF, so this is another issue.

When the DXF file opens fine in TrueView or [Autodesk Online Viewer](https://viewer.autodesk.com/), there is no need to create a new DXF. Then we definitely have an issue in `libdxfrw` with binary DXF and should investigate it with your file if possible.

I'm really eager to have a look at this binary DXF, as already said, it would be the first I ever seen. And I'm not sure if the author of `libdxfrw` ever tested binary DXF.
If it's not a problem, you can send the DWG too, I can add it to my private test collection for `libdxfrw`. But for this issue it's irrelevant by now.

If you need a fast solution, you can try to request a new DXF in ASCII format and see if that works. DXF 2010 should be fine, but ASCII is important as this part of the software is heavy used and better tested. Maybe it's worth to export in DXF R14 and 2000 ASCII format too. The older the format the better import should work, but possibly with less details.

lordofbikes, 2021-08-05:
Thanks for the file.
This is indeed an issue with `libdxfrw`, caused by code 310.
Code 310 is for binary data (e.g. preview image) which are ignored by LibreCAD anyway.
In ASCII DXF this is not a problem because the content is a hex string.
But in binary DXF the data following code 310 is a variable length field with binary data. These data can contain zero bytes, which are usually the end marker of strings. `libdxfrw` lacks a special treatment for this kind of data and assumes a string after code 310.
It stops reading at the first zero byte and this brings the further code/value reading of your file out of sync and following data is interpreted wrong.

The special format of code 310 binary data within binary DXF is not documented in DXF references I have. So it's not surprising to me, that this is not implemented yet.

Beside that, I'm not sure if you can use this file with LibreCAD at all. What I have seen from TrueView, the file contains multiple viewports and xrefs which are not supported. Some of the viewports even did not render well in TrueView, maybe 3D data?
So I'm not sure if a fix leads to a satisfying result with LibreCAD in the end.

dpocock, 2021-08-05:
Thanks for the detailed feedback

I'm not sure if it contains 3D data.  It contains 2D floor plans and 2D elevations of a building.

Would it be helpful for you to look at the DWG as well?

closed 2021-08-05 by lordofbikes

lordofbikes, 2021-08-05:
With my fix loading the binary DXF succeeds.

But as already mentioned, I'm not sure if LibreCAD is useful for editing.
Working is very slowly, freeze/unfreeze a layer needs about 2 seconds here.
This could be because of lots of entities or missing patterns.
The patterns are there, but not in LibreCAD format.
You can use Draft mode to disable the overlay image.
It seems that entities are grouped in viewports and blocks, I'm not sure what happens on edit and save.

I'm afraid that, after seeing the DXF, the DWG contains too much unsupported data for `libdxfrw`. DWG is a proprietary format and all available information comes from reverse engineering.
If you don't have worries, you can send the DWG for testing, it will not be used in public.

dpocock, 2021-08-06:
Thanks for the fast solution for that.  I pulled the changes and it also opens for me now.  I can view the layers of the building.

The architect who created the original file saved another copy as DXF for me.  It is ASCII DXF.  I can open that with the regular LibreCAD package on my system and also the latest version.

commit 415f8fcb (2021-08-05): fixed #1371, read failed with binary DXF