LibreCAD/LibreCAD #1373: Trim-tool entirely inactive when pressing <escape> button, not able to select second limiting entity.

Jochem043, 2021-08-12:
### Expected behavior
After you used the trim tool to trim one or more lines, you hit <esc>, and you are directly able to select another trimming entity.

### Observed behavior
After hitting escape, the trim tool is entirely not active anymore, so to use the trim tool you need to select it using command line, tool bar or menu. This makes the use of the trim tool less efficiënt.
 
### Steps to reproduce or sample file

### Operating System and LibreCAD version info
LC 2.2.0 RC2, macOS Big Sur 11.4

dellus, 2021-08-13:
Probably the behaviour was changed from 2.1.3 to 2.2.0 to avoid having to press ESC twice to end a command completely. This also applies to other commands.
Anyway, you can use the right mouse button instead of ESC.

lordofbikes, 2021-08-13:
This was indeed changed here: https://github.com/LibreCAD/LibreCAD/commit/7fbfe27f92182d6155b201fb142f6349151192cf
Means in 2.1.3 one has to push ESC twice to cancel several active commands and in latest version commands are canceled after first hit.
On the fly I found no discussion about this change, maybe it was a lonely one man decision.

Anyhow, from my using, I'd advocate for the current behavior. As mentioned by @dellus, my assumption is, that the reason for this change was the need of twice ESC to cancel operations.
For trimming, using the mouse is obligatory, so using right click to release current limiter and select another is more intuitive than using ESC I think.
For other commands, like 2-points-line, one may prefer the keyboard for input and then ESC would make more sense again.
Just checked this and found that in command line mode, when the command line is active, ESC is still handled like in 2.1.3.
When in _Specify next point_ state, ESC switches back to _Specify first point_ state. Only issue here, it leaves command line focus and only CTRL-M brings focus back to command line. Other triggers like SPACE or any other key, which usually activate the command line too, don't work after ESC.
This is another issue, I have to look if there is already an open one or if we have to create a new for this.

BTW, typing _escape_ to the command line is the same as pressing ESC in active command line, it also goes back from _next_ to _first_ point but keeps focus on the command line for next coordinates. Sadly there is no shortcut like _esc_ yet.
This also works for trimming, but this would be the worst work flow I can imagine.

Above mentioned issue in [LibreCAD/docs](https://github.com/LibreCAD/docs) is about ESC description in our manual, which currently is only the half truth and needs attention too.

Jochem043, 2021-08-17:
@dellus and @lordofbikes: thank you for helping out!

I didn't know about the right click option. Although the double-ESC action gives beginner level users like me the chance to pretty much stumble on the more efficiënt functionality (reusing the TRIM tool on a different entity without the need to reselect the tool), I agree with @lordofbikes and @lordofbikes that it gives unity to ESC button behaviour, while right-click works fine with me.

closed 2021-08-17 by Jochem043