LibreCAD/LibreCAD #1375: [Feature request] Edit text size along the top bar

MarcelloTheArcane, 2021-08-24:
When I run the `text` command, I can change the actual text and angle on the fly.

I'd like to also be able to edit the text height in the same way, rather than having to cancel the text command and re-run it in order to edit the size.

Here's the bar I mean:

[image]

### Operating System and LibreCAD version info

Version: 2.2.0-rc2
Compiler: GNU GCC 7.3.0
Compiled on: Dec 31 2020
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

fa201, 2021-08-24:
You can access to these data through **Properties** command to modify an object.

MarcelloTheArcane, 2021-08-25:
@fa201 That's true - that might speed up my workflow a bit.

I'd really like to adjust before I add the text though.

sand1024, 2026-09-26:
supported

closed 2026-09-26 by sand1024