LibreCAD/LibreCAD #1376: Librecad software shortcut keys

lianzaozi, 2021-08-24:
Does the librecad software have no shortcut keys?

MarcelloTheArcane, 2021-08-24:
LibreCAD has lots of shortcuts!  They are called commands.

Wiki: https://wiki.librecad.org/index.php/Commands

lordofbikes, 2021-08-24:
Or did you mean menu shortcuts?
[image]
Browse through the menus to find them, most are common used shortcuts.
Or press [Alt] and find the underlined menu letters to select menus and functions.

If you are missing any special shortcut, be more specific or ask in the [forum](https://forum.librecad.org). Experienced users often know better about using LibreCAD than devs.

Feel free to reopen, when you are missing a special shortcut and when you have added valuable information for a feature request.