LibreCAD/LibreCAD #1379: Plugins only loaded from `usr/share`

abenson, 2021-09-03:
Many distributions of Linux don't allow ELF files in `usr/share`;  Debian solves this by creating symlinks.  

I'd like to avoid symlinks and actually specify that plugins can be loaded from `usr/lib/librecad/plugins`.

Help? Thanks.

lordofbikes, 2021-09-03:
This is hard coded currently. But I touched this recently by chance for AppImage creation.
I'll consider this or maybe even add possibility to set this explicit in _Application Preferences_ in future.
Follow up e666fb172e3497dd00435ff435cc6d5177f3b8ee

u2fly, 2025-04-10:
@dxli

In latest `2.2.2_alpha` builds there are issues with default paths - LibreCAD does not detect distributed plugins location and "Plugins" menu is empty.

Same for fonts, templates, patterns and translation - it could be workarounded manually by setting "Paths" in preferences (but there is no field for "Plugins" path).

Obsered issue in latest AppImage and EXE builds (EXE tested in Wine on Linux).