LibreCAD/LibreCAD #1381: polyline not accepting command line coordinates

mMerlin, 2021-09-19:
### Expected behavior

Once the polyline tool is active, entering coordinates in command line text box should add nodes to the polyline

### Observed behavior

Coordinates entered in the text box are ignored. The command line prompt "Specify first point" does not change. The entered coordinate pair is shown in the command line text area, but no segment is added to the polyline.

entering lines through the command line work correctly.

### Steps to reproduce or sample file

in the command window
pl
0,0
1,1

The polyline becomes active, but the coordinates do not do anything. The text prompt remains at "Specify first point" and no segment is drawn in the main window.

Same result using
pl;0,0;1,1

### Operating System and LibreCAD version info

Version: 2.2.0-alpha
Compiler: GNU GCC 10.2.1
Compiled on: Jul 28 2020
Qt Version: 5.15.2
Boost Version: 1.73.0
System: Fedora 33 (Container Image)
5.13.16-100.fc33.x86_64

mMerlin, 2021-09-19:
The problem is not where I said. The symptoms were as described, but exit and restart librecad resolved it. Something must have gotten into a strange mode.

lordofbikes, 2021-10-03:
If this happens again, please have a look at the command line history.
Did there any messages appear when you hit enter?

I'll close this for now, it is not reproducible and it even works for you again.
You can come back and reopen it, if it happens again.

closed 2021-10-03 by lordofbikes