LibreCAD/LibreCAD #1386: Tangent 3 Circles tool fails randomly with parallel lines

lordofbikes, 2021-10-03:
This is a copy of SourceForge ticket [619](https://sourceforge.net/p/librecad/bugs/619/)

### Expected behavior
Using *Tangent 3 Circles* tool should provide 2 possible solutions when parallel lines are involved.

### Observed behavior
Randomly it is not possible to select the 3rd line.

### Steps to reproduce or sample file
* draw line
* draw parallel line
* draw intersecting line with parallels
* try *Tangential 3 Circles*
* select one of the parallels first

### Operating System and LibreCAD version info
Reproduced with latest code on Windows and Linux

Both samples on SourceForge worked well for me. My sample is from 1st SourceForge sample, the original test is on green layer *good*. The test that fails for me is on red layer *fail*.
**Rename** tangent3fail.txt to **tangent3fail.dxf**

[tangent3fail.txt](https://github.com/LibreCAD/LibreCAD/files/7274154/tangent3fail.txt)

lordofbikes, 2021-10-03:
https://github.com/LibreCAD/LibreCAD/blob/415f8fcbe4e8b256884c17da344b0fadc901a023/librecad/src/actions/rs_actiondrawcircletan3.cpp#L206-L216

This is the relevant place for this issue.
It is checked if selected lines 1 + 2 have an intersection or lines 1 + 3.
If there is no intersection, the lines are parallel and therefor swapped for further processing.
When the issue occur, an intersection between parallel lines is found by `getIntersection()`. This prevents swapping and leads to the issue in following code.

This is presumably a floating point precision issue which occurs only randomly, possibly related with line angles or coordinates.

So the question is why `getIntersection()` fails.
Is it possible to improve `getIntersection()` or must this issue be solved in the above quoted location?

melwyncarlo, 2021-11-27:
I could not reproduce this bug. I've tried several combinations. Maybe if a very distinct, step-by-step description is made. All three line endpoints would be useful to correctly simulate the bug.

commit c9ec48cf (2023-06-08): tangent 3 circle sometimes fails with parallels, fixes #1386 (#1440)

* Solved issue #1386

* Update rs_math.cpp: added comments

---------
closed 2023-06-08 by dxli