LibreCAD/LibreCAD #1403: "Closing drawing" dialog bug

CTuBu, 2021-11-13:
### Expected behavior
When click cancel on "Closing drawing" dialog return to current file and nothing else happened.

### Observed behavior
Current file closes and on next closing procedure ( file or program itself ) Librecad crashes

### Steps to reproduce or sample file
Close file ( from menu or "x" on filetab ) - Click cancel - Try to close other opened file or Librecad - Crash

### Operating System and LibreCAD version info
Win7 and Win10, LibreCad 2.2.0-rc2 from 15 to 23

melwyncarlo, 2021-11-16:
I don't see it.

lordofbikes, 2021-11-18:
follow up #1382, #1383

melwyncarlo, 2021-11-19:
I'm confused. If #1383 has sorted it out, why has it not been merged yet (since 23 Sep)?

I'll sort out my code anyway. It's a good learning experience.

melwyncarlo, 2021-11-19:
What about the other `slot` functions in the `qc_applicationwindow.h` file? 
I found four other `slot` functions bearing the return type `bool`.

melwyncarlo, 2021-11-19:
> follow up #1382, #1383

I have made the required changes.

And because I've used a default parameter, the same slot can also be called (from other files) 
_without_ using the second parameter.

lordofbikes, 2022-06-04:
The solutions #1382, #1383 and #1418 have all the same issue.
My comment in #1382 points to that issue, but was not totally clear or not the exact truth.

I had a deeper look now and I come to the conclusion, that this is basically a misconception to solve.
The reason to avoid slots with return values is, that in a multi threading environment a `emit()` doesn't call the slot immediately, but sends a signal to the event loop which calls the slot on its next iteration. And this issue applies to parameters of a slot method too.
Passing pointer or reference variables to a slot method does not guarantee that the parameter was processed after the `emit()`.
As mentioned in #1382, these solutions may work under certain conditions and surely have worked for the pull request authors. But under different conditions they may still fail.

imho the `accept()` or `ignore()` has to be decided in `QC_MDIWindow::closeEvent()` without forwarding any signal. I still don't have the whole oversight, with closing single MDI window or whole application with multiple modified documents. Needs some more investigations.

DeliveryFry, 2022-06-16:
I don't want to file a duplicate report.

So if I said:

1. When I click "Cancel" on the Close Dialog, my current drawing erroneously closes instead;
2. And right thereafter (with no drawings left open at all), click on the **Drawings** Menu;
3. LC immediately segfaults.

That's just another way of reproducing this _same_ bug, yes?

(Using **2.2.0-rc4**)

lordofbikes, 2022-06-16:
Yes, it's the same cause.

u2fly, 2025-04-18:
@dxli @@sand1024

Actual issue state:

1. Launch LibreCAD and change default `unnamed document 1` (without saving it)
2. File > Close
3. In a promt click "Cancel"

- latest 2.2.1_rc and 2.2.1.1 (release) — unexpetedly closes document (acting same as "Discard") — that is bug, fix should be backported for the next 2.2.1.2 (release) from 2.2.2.x
- latest 2.2.2.x — promts warning, but keeps document after closing warning:
>[!WARNING]
>Cannot auto-save the file
>Unnamed
>Please check the permissions.
>Auto-save disabled.

With changed, but earlier saved file (or opened existing file):
- latest 2.2.1_rc and 2.2.1.1 (release) — same bug as with unsaved default
- 2.2.2.x "Cancel" — works as expected (without issues)

sand1024, 2025-04-18:
@u2fly 
I've changed this logic locally, yet didn't commit it yet (in 2.2.2). The issue should be retested after next PR

closed 2026-09-26 by sand1024