LibreCAD/LibreCAD #1404: Separate the move and copy buttons

gerardbm, 2021-11-14:
### Expected behavior

The buttons to move and copy should be independent. This is how it works in the most CAD softwares.

The dialog should disappear. The move button should do simply a movement; fast and clean. The copy button should do a single copy by default, with the multiple copies value in the small input integrated in the statusbar.

### Observed behavior

They are one button with an annoying dialog.

### Steps to reproduce or sample file

It's an enhancement.

### Operating System and LibreCAD version info

All. It's an enhancement.

PD: Actually, rotate, scale and mirror could also be separate buttons when copies are needed. But the most important is "Move / Copy". I know this would be a big change, however this would increase the workflow in technical drawing. I can understand it's not easy to implement, since this was inherited from QCAD.

melwyncarlo, 2021-11-16:
Quite true. 

In fact, the drawings can be moved within the layer just by dragging them, 
and they can be moved into other layers by first selecting them and then just clicking on the other layer.

The _move_ operation button seems redundant. I personally think that the _Move/Copy_ button itself is redundant. 
The _copy_ operation can just as easily be accomplished by selecting the drawings, and then doing a _Control+C_ 
followed by a _Control+V_ (either on the current layer, or other layers). _Control+C_ would save it to the Clipboard.

gerardbm, 2021-11-16:
Keep in mind that one should have the ability to move a selection from one point to another point. The same for the copy functionality. This ability is not always achievable by dragging a selection of lines neither doing a <kbd>Ctrl</kbd>+<kbd>c</kbd> and <kbd>Ctrl</kbd>+<kbd>v</kbd>.

This three sentences should be true:

From one point to another point without additional keystrokes.
From one point to another point without having to close dialogs.
From one point to another point making use the snap tools.

This is how AutoCAD, Solidworks, Allplan, Microstation and ArchiCAD work.

melwyncarlo, 2021-11-16:
Isn't drag-and-drop already implemented? In the development version it is. All three sentences of yours are compatible with its functioning. This is regarding _move_, and NOT _copy_.

What do you mean by: `This ability is not always achievable by dragging a selection of lines neither doing a Ctrl+c and Ctrl+v.`

Do you mean that it is a challenge to implement, or do you mean that it's not what other CAD software do?

gerardbm, 2021-11-16:
The question is: why would you copy or move a selection with «drag and drop»? How do you use it for drawing? Show me an example, please. It seems you have never used CAD software.

We don't need to do strange things. Just split the copy and move button into two buttons, and remove the dialog.

melwyncarlo, 2021-11-16:
This video illustrates the drag-and-drop move feature that I was talking about:

https://www.youtube.com/watch?v=9vV4yupUMSw

Look at how the author of that video moves his drawings around. So, that's already been implemented.

gerardbm, 2021-11-16:
I will take a look at this video later. I have to go to work in 3 minutes. BTW, I think I will do a series of videos showing some interesting things. See you later.

melwyncarlo, 2021-11-16:
No problem; I would be eager to see them. See you later, and take care.

gerardbm, 2021-11-16:
Hi, @melwyncarlo. I'm back!

I'm sorry, but I didn't see a "drag and drop" action in this video. What he does is a normal usage of the function "move and rotate". From one point to another point, and rotate X degrees. And this is the right way to do the things.

For me, a "drag and drop" would be when you have a selection, and then you press the left mouse button, hold the left mouse button  while moving the pointer to a different position. That would be a very bad way of doing things. Why? Because you lose the ability of using most of the snap points. Try it: draw one horizontal line, one vertical line, activate "snap middle" and "drag and drop" from the center of the first line to the center of the second line, and you will see how you never draw a cross. With the normal usage (by clicking from one point to another point) you will be able to indicate from the middle of that line to the middle of the other line. And that's very important. Snap buttons are more than 50% of the productivity while drawing. That's why "drag and drop" is not even an option, despite CAD programs also do that, somehow.

Doing a "copy and paste" (ctrl+c and ctrl+v) is also a bad idea, because you would have still have more problems. With that, you can't do the movement "from one point to another point", since you can't indicate the first point, only the second point. You can't use the snap buttons. And you make use of the OS clipboard; never do that for drawing. The copy functionality is a very common task, we do it hundreds of times in a job of 8 hours/day, and we use the OS clipboard for OS operations with files or text. Actually, as a GNU/Linux user and programmer, in Vim we use the registers instead of the clipboard, in the terminal we use the primary buffer, and LibreCAD already has its own primary buffer, so there's no need to use the OS clipboard.

Going back to the main subject of this issue: when you use the button "Copy / Move" you have to do too many steps to do one action. It shouldn't be that way. It's slow. And what it's really very very slow, more than having to press the Ok every time, is to have to change this options when you want to switch from copy to move or viceversa, or when you have to indicate multiple copies. The image:

[image]

I'm not prettending that LibreCAD becomes a clone of AutoCAD, I'm prettending to make LibreCAD usable professionally. I woulnd't be here opening an issue if I hadn't enough drawing experience (since 2002). AutoCAD R14 (1997) was more productive than LibreCAD is in 2021.

GSStnb, 2021-11-16:
You can do the movement "from one point to another point" with Cut/Copy&Paste.  C&P requires two reference points, one for each operation:
  - The first reference point when copying / cutting the entity(ies) to define the where to cut/copy from.
  - The second when pasting the entity(ies) as a target.
Both sides of the operation will use any of the snaps; when defining the "source" entity(ies) reference point and then again for the target point.
The complete operation is:
1. Select entities.
2. Press [Ctrl]-[C] (or [Ctrl]-[X]) and set source reference point using snaps, absolute reference, etc...
3. Press [Ctrl]-[V] and set a target reference point, again using snaps, absolute reference, etc.  Repeat as needed...

gerardbm, 2021-11-16:
You are right, @GSStnb. I did the steps you mentioned and that worked for me. However, there are other reasons to don't do that, as I also pointed in my last comment. Using the clipboard and pressing so many times the copy and paste shortcuts is not a good workflow (you have to use the mouse and the keyboard and two hands for an action that should require only some mouse clicks); not so good as having a fast independent button for copy or move, with the minimal clicks. And that's the point of this issue, not finding alternatives with the same problem of the original button (a slow workflow).

gerardbm, 2021-11-16:
> It is fast - very very fast...

:)

Not so fast as it should be. Actually it's even worse than the original button to "Copy / Move".

gerardbm, 2021-11-16:
Look at this, what you do with copy and paste:

1. Select an entity (1 mouse click)
2. Press Ctrl+c or Ctrl+x (2 keystrokes)
3. Click an origin point (1 mouse click)
4. Press Ctrl+v (2 keystrokes)
5. Click an end point (1 mouse click)

So you are alternating mouse, keyboard, mouse, keyboard, mouse. Result: 3 mouse clicks and 4 keystrokes. It is very cumbersome. 7 actions in 5 steps alternating mouse and keyboard!

What we do with the "Copy / Move" button:

1. Select the entity (1 click)
2. Click the icon (1 click)
3. Click an origin point (1 click)
4. Click an end point (1 click)
5. Click Ok if it's good (1 click)

5 mouse clicks in a row, without using the keyboard. 5 actions in 5 steps! If it's not a copy, you have to read the options menu and check the move option, then we have to do 6 mouse clicks. 6 actions in 6 steps!

Now let's talk about what I'm proposing:

1. Select the entity to copy (1 click)
2. Click the icon (1 click)
3. Click an origin point (1 click)
4. Click an end point (1 click)

4 mouse clicks in a row! From one point to another point, and nothing else. You wanted to copy: so you copy a selection in 4 steps. You wanted to move: so you move a selection in 4 steps. That's how it should be.

GSStnb, 2021-11-16:
You are discounting the need to move out of the drawing area when using mouse with toolbars/menus.  Many CAD operators prefer to use the keyboard for selecting commands as it is quicker than using a mouse.  None the less, each to their own...

gerardbm, 2021-11-17:
> You are discounting the need to move out of the drawing area when using mouse with toolbars/menus. Many CAD operators prefer to use the keyboard for selecting commands as it is quicker than using a mouse. None the less, each to their own...

You said it well: «many CAD operators prefer to use the keyboard **for selecting commands**». And that's in the beginning of the action for the cases where you can combine the hand of the keyboard with the hand of the mouse. It's a reasonable combination. What it's not reasonable is to having to combine mouse, keyboard, mouse, keyboard and mouse to only move an entity or a selection. It is not an acceptable case in any case, unless you think you can work 16 h/day. Let's try to simplify the workflow!

We can take a look at some other CAD programs. QCAD has shortcuts composed by two letters to **start** commands. I mark the word "start" because it makes a big difference: you had the second hand on the keyboard and you started the command faster than pointing the mouse to an icon. And that's also fine! Still I think QCAD has a bad implementation: too many shortcuts composed by two letters that require the two hands to type them! The same happens with AutoCAD and its floating input box with an autocompleter, where you will need the two hands to autocomplete a lot of commands. The winner here is Blender, which has shortcuts of one letter (g to move, r to rotate, etc) to start actions very fast (it's not a CAD software, though).

The example you described with the copy and paste solution is the most long path to achieve, because you have to use the mouse, then the keyboard, then the mouse, then the keyboard and then the mouse again. Man! You never worked as a CAD operator, and sorry for this ad hominem. I've been working on it since 2002.

Let me tell you one thing about the workflow. Do you know why a lot of programmers use Vim or Emacs (but mainly Vim)? Because using the mouse is a waste of time when you have the two hands in the keyboard. Because programming **is a task done with the two hands in the keyboard**. For that reason, programmers also use some tiling window managers like i3wm, xmonad, bspwm, dwm with vim-like navigation. For the same reason, a lot of command line tools use the vim-like navigation (see cmus, ranger, vifm). For the same reason, there are plugins for some web browsers to use the vim-like navigation (vimium, cVim); or even exist web browsers with this navigation integrated (qutebrowser, luakit)...

In the same way, technical drawing **is a task done with one hand in the mouse and other hand in the keyboard**, trying to do the shortest way. If you have to use the two hands to type a command, then it's slow. If you have to do extreme combinations (the copy paste is one), then it's slow. The normal workflow in this cases is to use the second hand to enter numbers or start actions when it's faster than going to click an icon. 

LibreCAD could also add shortcuts to start actions, then we would have a killer fast path:

1. Select the entity to copy (1 click)
2. Press the imaginary shortcut to copy (let's say: **c**)
3. Click an origin point (1 click)
4. Click an end point (1 click)

4 actions i 4 steps. **Why this path would be the faster? Because you can start the action while you start moving the mouse pointer to the origin point. Without sacrificing the clipboard.**

lordofbikes, 2021-11-21:
Closing this means only that this feature will not find it's way into V2, but maybe found in V3 in future.
When time comes, we'll transfer these V3 tagged feature to V3 repo, don't duplicate them on V3 yet, this will only produce more work for us.

closed 2021-11-21 by lordofbikes

melwyncarlo, 2021-11-27:
I wasn't in my right mind earlier, so I couldn't quite grasp the simplicity of your question. Now, I see it, and it's already been implemented as simple Copy (`Ctrl + C`) and Paste (`Ctrl + V`) functions, without the options.

Check out my video :  [https://youtu.be/JC10MPmHdCs](https://youtu.be/JC10MPmHdCs)

gerardbm, 2021-11-27:
> I wasn't in my right mind earlier, so I couldn't quite grasp the simplicity of your question. Now, I see it, and it's already been implemented as simple Copy (`Ctrl + C`) and Paste (`Ctrl + V`) functions, without the options.
> 
> Check out my video : https://youtu.be/JC10MPmHdCs

Yep, it's what @GSStnb commented. As I can see now, it's not using the clipboard, what is really nice. 

However, one problem I'm finding is: if I want to cut one entity from one point and I want to paste it to another point of the same entity, the element disappeared (because I did a Ctrl+x), so I lose the point at the moment of pasting it.

Offtopic: by the way, I'm enjoying a lot the fixes you are doing these days. Congrats for the good work you are doing.

melwyncarlo, 2021-11-28:
Thank you very much @gerardbm! :smiley:

lordofbikes, 2022-04-28:
Evaluating #1438 I scanned over this again.
Just to recognized, there are already cut/copy/paste icons without a dialog.
Don't use the Move/Copy icon from the left tool bar, but one of the cut/copy/paste icons in the top `Edit` tool bar. Then copy/paste is just 5 clicks, no keyboard and no dialog.

gerardbm, 2022-04-30:
Thanks for evaluating this. It would be a good idea if we could customize the toolbar, so we could create two buttons (one that does a Ctrl+C and Ctrl+V mixed and one that does a Ctrl+X and Ctrl+V mixed). Then I would hide the button Move/Copy. That would be interesting.

xieve, 2023-09-12:
Ctrl-X Ctrl-V behaves very badly with blocks and thus isn't a good alternative to `mv`: It empties one block and creates a new one with `-0` suffixed. When I try to delete the original, now-empty block and rename the new, copied one to the original name, it says that a block with that name already exists. I second the original motion of this issue, also because that is the workflow most people are used to from other CAD solutions.