LibreCAD/LibreCAD #1405: Apple M1 support

Olegt0rr, 2021-11-15:
As described in [release](https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc2) latest (2.2.0-rc2) version has M1 support, but it's not true. 

The app successfully installed and launched, but via Rosetta (Intel)!
[image]

PeterHagen, 2022-12-18:
With the 2.2.0 release, there is no M1 version available unfortunatly

lordofbikes, 2022-12-18:
M1 packages, which were generously provided by a community member, were reported as not running. Even when it turned out, that it's probably only a matter of signing, we stopped that to avoid further issues.
I close this in favor of #1548.

closed 2022-12-18 by lordofbikes

PeterHagen, 2022-12-18:
> M1 packages, which were generously provided by a community member, were reported as not running. Even when it turned out, that it's probably only a matter of signing, we stopped that to avoid further issues.
> I close this in favor of #1548.

Yeah, that was me. The second one I provided did work, and I am actually still using it.

lordofbikes, 2022-12-18:
Sorry @PeterHagen, I didn't link that together.
Because of the discussion in #1548 I didn't ask you for a 2.2.0 release package.
Maybe in the new year, please see my recent response in #1548.