LibreCAD/LibreCAD #1420: Add button to break out segment

gerardbm, 2021-11-18:
Enhancement to add a button to break out a segment limited by intersecting entities.

This button is already implemented in QCAD. See the relative [article](https://qcad.org/doc/qcad/latest/reference/en/scripts/Modify/BreakOut/doc/BreakOut_en.html).

[Video demo](https://www.youtube.com/watch?v=oivRKom4ri8).

melwyncarlo, 2021-11-19:
Do you mean something like [this](https://forum.librecad.org/Cut-Line-or-Trim-Between-Two-Intersections-td5706891.html)?

gerardbm, 2021-11-19:
Yes, it's the same function. QCAD and Allplan have this function well implemented and it's awesome: with one click one can cut segments continuously; as I show in the video.

The solutions provided by _sp1der_ and _cantcode_ are not productive. It's faster making copies and snaping to intersections than their recommendations.

melwyncarlo, 2021-11-19:
Sounds interesting. It sounds similar to the solution I plan on implementing (if approved) for issue #1340.

If I successfully implement that, then this could be considered an extension to that solution.

gerardbm, 2021-11-19:
Yes, this would be nice. You can also check QCAD to see the idea more in details. It's so useful that we use this function hundreds of times every day in my office (via Allplan). A lot of segments limited by one or two entities, and you cut them one by one only clicking in the segment you want to cut. 

In my video there's the problem that the mouse pointer is invisible. Every segment that desapears is the result of one click in this zone.

Thanks!

dellus, 2021-11-20:
Hey melwincarlo! If you could get this one-click-cut-out-section feature done it would be a really great contribution. A tremendous time-saver. Eagerly awaiting....

dellus

lordofbikes, 2021-11-21:
Closing this means only that this feature will not find it's way into V2, but maybe found in V3 in future.
When time comes, we'll transfer these V3 tagged feature to V3 repo, don't duplicate them on V3 yet, this will only produce more work for us.

closed 2021-11-21 by lordofbikes

melwyncarlo, 2021-11-23:
I'm wondering (with regards to that video), after the trimming operation, does it all remain a combined polyline entity, or do they separate: either as discrete segments, or as broken polylines?

Secondly, wouldn't it be better to create a tool that converts a polyline into discrete segments (at all the intersections)? 
You could then just delete any individual segment as you please. 
There already is a `Create Polyline from Existing Segments` tool.
How about a `Discretize` tool?

dellus, 2021-11-23:
In gerardbm's video there is probably no polyline but just a sequence of ordinary line entities. I have been using Allplan (having such a tool) for many years and must confirm it is very,very useful that way.
To make this tool applicable on a polyline would be an additional extension.
A tool that converts a polyline into discrete segments at all intersections is a possibility, but I think useful only for special cases. Most times there will be the need to cut out segments at certain places only, not at all intersections.
What does "Discretize" mean? You can "Explode" a polyline into single entities in Tool - Modify.

gerardbm, 2021-11-23:
Exactly, it is what @dellus comments. Cutting ordinary lines is a very very useful functionality and its usage becomes a common task. In Allplan, when one tries to break out a segment of a polyline, this tool breaks a polyline into two polylines after removing the clicked segment.

Also, +1 about this:
> «A tool that converts a polyline into discrete segments at all intersections is a possibility, but I think useful only for special cases.»