LibreCAD/LibreCAD #1421: Add button to snap middle manual

gerardbm, 2021-11-18:
Enhancement to add a add a new method to snap to a point that is in the middle between two points.

This snap is implemented in QCAD. See the relative [article](https://qcad.org/doc/qcad/latest/reference/en/scripts/Snap/SnapMiddleManual/doc/SnapMiddleManual_en.html).

[Video demo](https://www.youtube.com/watch?v=4pGMmmIxgbI).

melwyncarlo, 2021-11-19:
What about the existing **Snap middle** button that does the same thing?

gerardbm, 2021-11-19:
_Snap middle_ is also useful to snap to the middle of an entity. However, this functionality (also well implemented in QCAD and Allplan) allows you find manually a middle point between two points without having an entity there. It's useful for the origin point and the end point.

Actually, old QCAD versions didn't have this function, so I was surprised when I saw this in recent versions. Since QCAD and LibreCAD are "brothers", I think it shouldn't be difficult to implement.

melwyncarlo, 2021-11-19:
Not that challenging. I can get on with it once its approved as a _feature request_.

gerardbm, 2021-11-19:
Yeah, that would be super nice! 

BTW, I should learn C++ to help making contributions. At the moment I have I compile the source code, so from time to time I do a 'git pull' to check the improvements.

melwyncarlo, 2021-11-20:
Done and tested! I'll upload my code soon.

melwyncarlo, 2021-11-20:
> BTW, I should learn C++ to help making contributions. At the moment I have I compile the source code, so from time to time I do a 'git pull' to check the improvements.

I would suggest learning from the maker itself :  The C++ Programming Language by Bjarne Stroustrup
I gave away that book, and so now I refer to :  C++ Primer Plus by Stephen Prata

There are many online resources too: [learncpp.com](https://www.learncpp.com/), [cplusplus.com](https://www.cplusplus.com/), [TutorialsPoint](https://www.tutorialspoint.com/cplusplus/index.htm), [W3Schools](https://www.w3schools.com/CPP/default.asp), etc.
There are many free e-books too.

melwyncarlo, 2021-11-20:
The best part about building from source is that you get to make mistakes and try again and again till you succeed, and learn a lot on the way.

melwyncarlo, 2021-11-20:
> Done and tested! I'll upload my code soon.
> 
> https://youtu.be/XZU929ykxvg

In the video, the dotted lines seemed too faint, so I changed them to dots-and-dashes, as seen below:

[image]

melwyncarlo, 2021-11-20:
You can replace the appropriate files in your build folder, and test the tool.

But remember to first delete the *.conf files in the following directory :  `$HOME/.config/LibreCAD`
The files on my computer were named `LibreCAD.conf` and `LibreCAD1.conf`

lordofbikes, 2021-11-21:
@melwyncarlo it is not good to advice people to delete their configuration file, where all the GUI settings are stored.
The file is restored on next start, but with default settings. This will destroy the users settings and may need much work for him to restore his familiar environment.
Also this only applies for Linux, maybe for Mac OS too, but surely not for Windows.
And at least, the configuration should not influence testing of self built versions.

melwyncarlo, 2021-11-21:
Oh, I'm truly sorry, I didn't realize that. Perhaps if the user made a copy of their configuration file beforehand; 
though I suppose it would be risky for those who are not aware about changing it back correctly.

I suggested that because, strangely, for my tool to show up on the toolbar, 
I had to delete the configuration file, and only then did it show up.

The tool itself works fine without configuration, and also shows up in the menu.

lordofbikes, 2021-11-21:
Concerning this feature request and the resulting pull request #1424 I need to say this:
I understand the motivation of @melwyncarlo to get his hands dirty on this and I appreciate your thirst for action.
But I already warned on Zulip, that this may be more complex than it seems. This has been confirmed and discussed in #1424.
Many thanks for your efforts. I haven't looked detailed in this code, but I'm sure it was good training to get familiarized with the code base.
Anyhow I do not recommend to continue this work.

We are trying for years now, to find a final version for V2 and concentrate forces on V3. Sadly these forces reduced to a minimum over past years. When we don't stop to implement new features, we will stick with V2 and QCAD license dependence forever.
We need to tag this kind of features with a V3 label soon and close them to avoid that anybody else puts efforts into them in this code base.

So closing this means only that this feature will not find it's way into V2, but maybe found in V3 in future.
When time comes, we'll transfer these V3 tagged feature to V3 repo, don't duplicate them on V3 yet, this will only produce more work for us.

lordofbikes, 2021-11-21:
The simple workaround for those missing this feature is to draw a simple _2-Point_ line an use existing _Snap on Middle_. After the related action the line could be deleted again or put it on a dedicated layer, which could be hidden later.

No need to discuss this, I understand, that there are power users, who find fault with this workaround for a few clicks more.
But this is how it works now.

closed 2021-11-21 by lordofbikes

lordofbikes, 2021-11-21:
> Oh, I'm truly sorry, I didn't realize that. Perhaps if the user made a copy of their configuration file beforehand; though I suppose it would be risky for those who are not aware about changing it [...]

@melwyncarlo when anything is obscure, don't hesitate to ask. As long as you pushed your code to your fork, I can pull it and have a look before you start a pull request. Then we can eliminate issues before your pull request gets attention in the main repo.

melwyncarlo, 2021-11-21:
@lordofbikes Thank you very much. I'll restrict complicated solutions to checks before pulling requests.
Yes, I agree, asking is better than doing something wrong.

I did familiarize myself with the codebase quite a lot. I think, apart from minor fixes and feature requests in V2, I'll move towards V3.

gerardbm, 2021-11-22:
Congrats, @melwyncarlo. Your implementation is more than perfect, because this option to add the percentage is also really useful, and I use it everyday in Allplan via its snapping button called [division point](https://help.allplan.com/Allplan/2021-0/1033/Allplan/10643.htm). Your implementation is even better, because if I want to find a snap point in a line divided in 4 sections, I will add 25%, 50% or 75%, and if I want to find a snap point in a line divided in 3 sections, I will add 33.333333% or 66.666666%. A lot of possibilities! Nice, thanks a lot!

melwyncarlo, 2021-11-22:
Thank you very much, @gerardbm! I'm glad you liked it. :smiley:

dellus, 2021-11-23:
@ gerardbm, instead of 33.333333 you may also type 100/3, so you don't have to bother accuracy.