LibreCAD/LibreCAD #1428: Importing a 2D DXF file generates Z-axis scaling of 0 for DXF output

egilkv, 2021-11-23:
### Expected behavior
Correct DXF file generated with Z scale of non-zero.

### Observed behavior
Having done "Save as" to a DXF file, a 3D tool like FreeCAD chokes on the DXF input since the Z scale in the input (param 43) is zero.
Error is in line 3250 of test-cu.dxf
It only seems to be a problem when the DXF file is read through an import. 
Note that the imported file (that came from KiCAD) has an X scale (parameter 41) of 1, and Y scale (parameter 42) of 1, but no Z scale (parameter 43), which by DXF spec should default to one.

### Steps to reproduce or sample file
File -> New
File -> Import -> Block -> test-F_Cu.dxf
Add one item at reference point 0,0
File -> Save as -> test-cu.dxf

### Operating System and LibreCAD version info
OS: Ubuntu 18.04.6 LTS
Version: 2.2.0-alpha (but also earlier releases)

egilkv, 2021-11-23:
Sample files:

[test-F_Cu.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/7588817/test-F_Cu.dxf.gz)

[test-cu.dxf.gz](https://github.com/LibreCAD/LibreCAD/files/7588820/test-cu.dxf.gz)

lordofbikes, 2021-11-23:
This sounds somehow familiar, but I couldn't find another issue or probably a fix with a quick search, neither here, nor in https://github.com/LibreCAD/libdxfrw.
The 2.2.0-alpha label is used, when sources are not pulled by git or pulled without tags.
Can you specify your version better? Downloaded or built from source? Optimum would be a git hash.

egilkv, 2021-11-23:
I have pulled the latest source, and reproduced the issue there. Details are:

Version: 2.2.0-rc2-24-g1d314270
Compiler: GNU GCC 7.5.0
Compiled on: Nov 23 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.6 LTS

I do believe the bug can be fixed by changing line 456 of librecad/src/lib/modification/rs_modification.cpp from:

    RS_Vector vfactor = RS_Vector(factor, factor);

to:

    RS_Vector vfactor = RS_Vector(factor, factor, factor);

However, the treatment of the zscale is very lax generally, and since this is output in the resulting DXF, there are also other instances that could end up with a zscale of zero. I will try to provide a pull request.

egilkv, 2021-11-23:
Proposed fix is in https://github.com/LibreCAD/LibreCAD/pull/1429

egilkv, 2021-11-24:
With regards to this patch, someone that knows these things should verify if a Z scale of zero in the output is valid DXF or not.