LibreCAD/LibreCAD #1434: Aren't freeze and hide visibility supposed to be the same thing?

melwyncarlo, 2021-11-26:
The below two images illustrate the supposedly inaccurate distinction.

<hr>

[image]

<hr>

[image]

<hr>

So, toggling the visibility of an individual layer, by clicking on a given _layer's eye icon_, 
and freezing all, by clicking on the _layer list's eye icon_, 
points to different functions.

I wonder why. `frozen` is one member variable, and `visibleInLayerList` is another.
Isn't one of them just redundant?

They share the same _eye icon_. They've got to share the same functions too!

In the doc, it says `Show all layers`, but in the software's tooltip, it says `Freeze all layers`. Confused.

This is with regards to the development version (2.2.0) only.

Guruprasad-Rane, 2021-11-26:
Freeze and Hide are not same.

When a layer is frozen, blocks on that layer get totally hidden.

When the layer is hidden, only entities inside block with that layer get
hidden. Entities on other layers are still visible.

Hope this helps. If required I can explain in further detail.

Anyways good to see activity in the community. Keep up the good work.

Regards
Guruprasad

melwyncarlo, 2021-11-26:
@Guruprasad-Rane Thank you very much! :smiley:

You are right; as per the docs, `Show/Hide` is separate, and `Freeze` is synonymous with `Lock`.

Then why is the eye icon, which should only deal with `Show/Hide`, perform the `Freeze` function?
That is my actual question. 

When I toggle visibility to `false` (hide), and then check visibility via the `isVisibleInLayerList()` method, 
the function returns `true` (visible). 

This is because, in the code, when toggle visibility function is called, the function changes the layer's 
**frozen** state, and _NOT_ the **visibility** state (which it should actually do).

I've solved it now, I'm wondering if I should pull it. I'll do so, if it is officially declared as a bug.
What I did was that I dealt with the visibility state upon freezing.

Note that the `lock` function (which as per the docs, is freeze) is again, a completely different function.

lordofbikes, 2021-11-26:
The `freeze` term comes from AutoCAD. I don't know if and how they use it in their GUI in past or today. At least this is how the visibility property is called in DXF reference even when it's kind of misleading. Anyhow, I think this is common word usage in AutoCAD universe and may confuse CAD beginner.

Basically `freeze` means the visibility of a layer. So when it is frozen, all entities on this layer are hidden.
I'm not sure to confirm @Guruprasad-Rane's explanation, if this is somehow related to blocks.
It is as far as the `INSERT` could be on a different layer than the `BLOCK` content. When the layer of the `INSERT` is frozen, the whole block is hidden. But when a block uses multiple layers and one of them is frozen, then only the entities on this layer are hidden, the other block entities stay visible.

Concerning the manual, this should be addressed to rework in the manual at https://github.com/LibreCAD/docs, with reference to this issue. `Freeze` is the term to show/hide layers, connected with the eye icon and `Lock` is the term to protect entities from being selected or modified and is connected with the padlock icon. Locked layers stay visible.

The source of confusion in programming is the misinterpretation of the source code. `isVisibleInLayerList()` has nothing to do with layer properties, especially `freeze` property. As the method name says, it only returns if the **layer is visible in the layer list**, but not if it is frozen.
There is a filter box in the layer widget which allows to filter layers by name. This makes layers invisible in the layer list, no matter about their `freeze` state. And the `freeze all` icon is intended to modify only layers which are visible in the layer list and keep layers untouched which are invisible in the list.
With that said, I must assume, that the related #1433 is not a valid solution.

melwyncarlo, 2021-11-26:
Oh, yes, the filter box. That makes sense! Thank you very much @lordofbikes 

My question had got nothing to do with blocks (atleast not directly), just layers.

So, my previous pull is nullified. I am now using the `isFrozen()` method instead. 
I agree that the term is misleading. 

Upon further inspection, it turns out that the `visibleInLayerList(bool)` set method
is called from the `qg_layerwidget.cpp` file, which uses it internally to 
hide the non-relevant **layer details** (or **layer widgets**) in the layer list, 
as the user types into the input filter textbox.

I have already posted an issue in the Docs repository with regards to the naming.

closed 2021-11-26 by melwyncarlo

GSStnb, 2021-11-26:
The manual follows the software.  "Freeze" / "Defreeze" are used in the context menu (right-click on a layer in the layer list)  file:///home/garys/Downloads/dock-layerContextMenu.png.  It should be removed from there and the manual will follow suit.

lordofbikes, 2021-11-26:
Many thanks @GSStnb, I haven't checked this with my replies.

Now I did and `freeze` is also used in block list to show/hide blocks.
Also changing sources, will break all translations which are already done.
So changing source doesn't make sense for me.

fa201, 2021-11-26:
@lordofbikes @feragon Would it make sense to change the wording for V3 ?