LibreCAD/LibreCAD #1439: Not all entities displayed while dragging them to new position

jbergengh, 2021-11-28:
After selecting a number of entities, while dragging them to a new position, only a small number of them are displayed.

### Expected behavior
All entities should be displayed while dragging, or perhaps a bounding box if there are a tremendous number of entities.

### Observed behavior
See short video for example

### Steps to reproduce or sample file
Use a selection box to select the entire drawing.
Left-click drag the group to another position.
Observe the entities while dragging.  Usually only a subset of entities are displayed while dragging.  Sometimes all are, sometimes very few are, depending on the drawing.

### Operating System and LibreCAD version info
Linux Mint 20.2
LibreCAD version 2.1.3, GNU GCC 9.3.0, Qt 5.12.8, Boost 1.71.0
LibreCAD version 2.2.0-rc2-20, GNU GCC 9.3.0, Qt 5.12.8, Boost 1.71.0

https://user-images.githubusercontent.com/55407804/143730788-69129d62-7740-410d-a72b-914fd05d709f.mp4

[test.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/7612857/test.dxf.zip)

melwyncarlo, 2021-11-28:
Go to `Options → Application Preferences → Graphic View → Number of preview entities`

_Number of preview entities_ is actually the _Maximum number of preview entities_.
Increasing its value should do the job, but 800 seems to be the maximum a user can set.

Be aware that while a polyline (consisting of _N_ lines) is considered as one entity, 
an apparent polyline (made up of N individual lines) created by the _2 points_ tool 
would be considered as selecting N individual entities.

So, if you add them all up, and it's more than the maximum setting, then 
some of it will not be shown.

melwyncarlo, 2021-11-28:
At your own risk, if you wish to break that rule, you could make the following changes to file `LibreCAD/librecad/src/lib/actions/rs_preview.cpp` at line number `41` :  

Change from `maxEntities = RS_SETTINGS->readNumEntry("/MaxPreview", 100);`
to `maxEntities = 1000; //RS_SETTINGS->readNumEntry("/MaxPreview", 100);`

The value `1000` may be replaced as required. It is good practice to just comment out the original code while experimenting, in order to easily revert back, if desired.

It is also important to note that _risks_ include LibreCAD crash and computer hang/freeze.

jbergengh, 2021-11-28:
Thank you.  Even a small, simple drawing like this quickly exceeds the limit (set to 800 in this example).  I suspect it is the text - maybe every segment of every charactert counts an an entity.  Perhaps not including text and splines be considered?

https://user-images.githubusercontent.com/55407804/143778251-848d9986-d4e2-4989-a642-40499db6d12b.mp4

jbergengh, 2021-11-28:
@melwyncarlo I did some testing.  I created a 40,000 entity drawing and changed maxEntities to 20,000.  Other than some slowness, I had no problem moving geometry around extensively.  RAM usage was flat at 2.3gb throughout the testing and one of my 4 cpu cores would hit 100% only while moving geometry.  No crashes or lockups.  Dell laptop with 8gb ram, core i7 cpu@3.1ghz.

What would you think about adding an option for a user-defined value for the Number of Preview Entities?  Perhaps with an upper limit of maybe 10 or 20,000 to keep the user from entering an insanely high value?  I think the current limit of 800 is far too low.

If you think that's not a good idea, then let's close this one out.

jbergengh, 2021-11-28:
I didn't realize that the user can enter any value desired under Options - Application Preferences - Graphic View - Number of preview entities.  No need to revise the source code.  I increased it all the way up to 20,000 entities, then tested it for quite a while with a 40,000 entity file, and I didn't experience any crashes or lockups.  Memory usage held flat at 2.3gb throughout and one cpu core out of 4 would go to 100% while moving entities, then drop back to normal.   This solves the issue imo, so I am closing this issue.

closed 2021-11-28 by jbergengh