LibreCAD/LibreCAD #1451: Grid appears when opening a second file

fa201, 2021-12-05:
### Expected behavior
If the *grid* is off (not shown) on the current file then opening a second file should not change the setting on first file

### Observed behavior
Opening the second file does change the *grid* setting on the first file.

### Steps to reproduce or sample file
Open a drawing with the *grid* off
Open another drawing or click on *New*.
Initial drawing has the *grid* but the new drawing has not.

### Operating System and LibreCAD version info
Version: 2.0.7-1810-geba14138
Compiler: GNU GCC 9.3.0
Compiled on: Nov 30 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.3 LTS

melwyncarlo, 2021-12-07:
I don't think that this is a bug. Note that the grid settings are located within the _Current Drawing Preferences_, meaning that the settings apply only to the _current drawing_, and _not_ to the other drawings.

jbergengh, 2021-12-07:
Try the following steps to reproduce:
1. Start LibreCAD
2. File - New
3. File - New
4. File - New
5. Drawings - Arrange - Tile Vertically
6. Select each drawing in turn and toggle its Grid Display off
7. Confirm none of the drawings are displaying a grid
8. File - New
9. Observe the newly created drawing has its grid display ON
10. Here is the bug -> Observe one of the previously created drawings also had its grid display turned on.

melwyncarlo, 2021-12-07:
I still don't see it.

[https://youtu.be/NcjNC8YlqHA](https://youtu.be/NcjNC8YlqHA)

jbergengh, 2021-12-07:
https://user-images.githubusercontent.com/55407804/145075228-cb00e897-85e6-47fc-8c2c-2f3b066c24eb.mp4

melwyncarlo, 2021-12-07:
Got it now! Turns out that the grid setting in the `Current Drawing Preferences` and the tool button are (though not meant to be) separate; they seem to do their own things. Upon disabling the grid in the preferences, the tool button does not get updated. This is clearly a bug related to `linking`. I'll fix it tomorrow morning.

fa201, 2021-12-07:
@jbergengh Thanks for the video, very clear.

commit 176fa877 (2023-06-24): grid appears in 2nd file, when it is disabled in 1st file, fixes #1451 (#1457)

* Solved issue #1451

* Update qc_applicationwindow.cpp

* Update qc_applicationwindow.h

---------
closed 2023-06-24 by dxli