LibreCAD/LibreCAD #1452: Mark relative position coordinates in status bar with @ (small GUI improvment)

dellus, 2021-12-05:
It is confusing to see absolute and relative coordinates running side by side in the status bar, at first sight I'm always not sure what is what. Newbees often don't realize the meaning at all.
So what about just putting an "@" in front of the readings. Seems simple, so maybe ok to include in 2.2.0. 
As i can't insert an image here I have posted it in the forum at Suggestions & Feedback.

fa201, 2021-12-06:
[image]
@dellus Well the '<' is for relative but adding a '@' would make more clear in my opinion. At least it would not harm...

dellus, 2021-12-06:
I know drag and drop should work, but it doesn't, whatever I try. Maybe it is because my old win xp, or the mypal browser.
I also know the '<' is for relative. I find it to look less confusing when visually grouped by the '@'s.

melwyncarlo, 2021-12-07:
Based on the above image (hoping it's the result of relevant code changes), I'd say that this issue has already been solved?

dellus, 2021-12-07:
Sorry, it's just the result of image manipulation.

melwyncarlo, 2021-12-07:
Oh, okay. So, it's up for grabs!

fa201, 2021-12-07:
My mistake. '<' means polar I was not in my mind...
[image]
Indeed a relative icon or character is missing.

commit 99208a0c (2022-06-06): Merge pull request #1456 from melwyncarlo/Task_40

mark relative coordinates in status bar, fixes #1452
closed 2022-06-06 by lordofbikes