LibreCAD/LibreCAD #1453: In Drawing Preferences - Dimensions - Fixed length change naming "units" to the actually used units like in the other fields

dellus, 2021-12-05:
Another tiny GUI improvement:
In Drawing Preferences - Dimensions - Fixed length the naming is "units" instead of the actually used units (mm, m, inches or whatever) like in the other according fields. This might induce the user to think something here works differently. 
This might have made some sense in V 2.1.3, as there the length did not submit to "General Scale". In 2.2.0 it does submit, so it behaves the same way as the other settings.
So why not also use the actual units naming?

All OS'es, V 12.1.3 and 2.2.0

fa201, 2021-12-06:
[image]
I agree

melwyncarlo, 2021-12-07:
[image]

closed 2022-06-06 by lordofbikes