LibreCAD/LibreCAD #1458: Refresh issue on custom color for layer attributes

fa201, 2021-12-12:
### Expected behavior
I use custom colors for layers. When clicking on *Modify layer attributes* icon, I should see the appropriate custom color related to the given layer.

In the *Layer settings* window, I would accept the custom layer default icon to appear for the given layer so showing that it is a custom layer but not showing the *actual* custom color.

### Observed behavior
I see the *Light gray* color for this given layer whatever custom color is chosen.
I use the last build which I compiled and launch with the console.
In the console I have the following message:
`qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 5537, resource id: 12774756, major code: 40 (TranslateCoords), minor code: 0`

I had screenshots of custom colors as they may be OS specific.

### Steps to reproduce or sample file
1/ Click on *Modify layer attributes* icon for a given layer.
2/  Define a custom color (blue for instance) for a layer and validate.
[image]
3/ The color is changed.
[image]

3/ Click on *Modify layer attributes* icon for the same layer. It shows *Light gray* whatever custom colors.
[image]

### Operating System and LibreCAD version info
Version: 2.0.7-1810-geba14138
Compiler: GNU GCC 9.3.0
Compiled on: Nov 30 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.3 LTS

MarcelloTheArcane, 2022-07-19:
I can reproduce this in the following version:

Version: 2.2.0-rc4-5-g18e8b09d
Compiler: GNU GCC 12.1.0
Compiled on: Jul  1 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

gerardbm, 2023-06-04:
I think this pull request should be merged if the solution is acceptable.

closed 2023-06-08 by dxli

fa201, 2023-06-09:
Thanks @melwyncarlo  & @dxli