LibreCAD/LibreCAD #1459: Elipse center with 3 points causes freeze when choosing second point.

howtopam, 2021-12-16:
### Expected behavior
Choosing center point of elipse with center and 3 points should not cause program freeze and data loss.
### Observed behavior
Attempting to draw an elipse with center and 3 points after clicking on center point with snap on intersection of polar coordinate isometric view drawing Librecad freeze before clicking second point. Error message read; "cannot have circle with 0.00 radius." Hit ctrl + c key to close Librecad application. Discovered upon re-opening Librecad autosave did not work because there was no #file.dxf backup in saved folder. Re-open Librecad 2.2 appimage and initial drawing.dxf file to discover last hour of work before freeze was lost.
### Steps to reproduce or sample file
Re-opend Librecad application re-opened file.dxf discovered much work lost. Re-draw some polar coordinate isometric view structure to where elipse could be drawn once again. Choose elipse with center and 3 points tool and licked on center point with snap on intersection selected and Librecad freeze once again. Hit ctrl + c key on keyboard to close Librecad application. Re-opened drawing in stable version Librecad 2.13 and could draw the elipse without issue. 

### Operating System and LibreCAD version info
Kernel: 5.11.0 x86_64 bits: 64 Desktop: Xfce 4.12.4 
           Distro: Debian GNU/Linux 10 (buster) LibreCAD-2.2.0-rc3-x86_64.AppImage

lordofbikes, 2021-12-16:
Finally found the correct error message `RS_Circle::createFromCR(): Cannot create a circle with radius 0.0.`
This is just an info and doesn't crash for me on Windows. It is triggered, when the mouse is moved over the center point again.
I can later try if this is related to Linux or AppImage somehow.

melwyncarlo, 2021-12-18:
I cannot reproduce the error on my Ubuntu version 20.04.3 LTS; perhaps a video could help.

In the file `rs_circle.cpp`, in function `bool createFromCR`, at line number `151`, the RS_DEBUG warning should not cause a freeze. It simply tests against a condition, and the condition causing the problem is `fabs(r) > RS_TOLERANCE`. It occurs when (pragmatically) `r = 0`, where `r` is the radius. 

There are no loops (to assume infinite looping) or memory issues.

Furthermore, this function is called from the function `bool preparePreview`, in the `rs_actiondrawellipsecenter3points.cpp` file, at line number `141`. So, according to the code, the value `0.0` occurs during the `mousemove` event, when the first point is placed _at the center point_ of the ellipse (which was selected previously via mouse click).

So, either the problem is emanating from another source, or the problem is specific to the app image being used.

howtopam, 2021-12-18:
Thank you for taking your time to look into this issue. I am going to guess that the problem is relative to the AppImage only. I did have another issue relative to the same issue in question. After I was able to create two indentical elipeses one located 7.5 mm below the other in the isometric view I selected the divide tool and selected the lower elipse and went to divide the lower elipse on the intersection of the upper elipse, but could not. Choosing snap on intersection and clicking on the intersection of the upper and lower elipsees the selected lower elipse would not deselect - indicating the lower elipse line had been successfully broken. I saved the drawing from within the Librecad 2.2.0_rc3.AppImage and re-opened the drawing file in my Librecad 2.1.3 install and I was able to successfully divide the lower elipse line. I will continue to use the Librecad 2.2.0_rc3.AppImage to see if the issues reoccur in future reproduced drawings. Thank you again for your time expeditious reply.

melwyncarlo, 2021-12-19:
I have just tested using the `Librecad 2.2.0_rc3.AppImage` file. It seems to work all fine; no issues. I have attached a video below demonstrating the same. Could you provide a counter-video?

<hr>

https://user-images.githubusercontent.com/66683108/146663722-19cadcc7-7ba0-4d39-81fa-2a6dd813d93c.mp4

melwyncarlo, 2021-12-19:
I, however, did face an issue with regards to snapping to intersection. As shown in the video below, the snapper never snaps to intersections, and so, upon clicking, the ellipse does not get _de-selected_. Instead, you should see the following error message in the command view: `Cutting point is not on entity.`

If this is your actual problem, then the problem is related to the `Snap Intersection` tool, and not the `Ellipse` tool. On my end, it seems to occur randomly; not yet sure if there's a pattern to it.

<hr>

https://user-images.githubusercontent.com/66683108/146663826-4e449406-cebd-4952-a16e-3bba9e5aee43.mp4

melwyncarlo, 2021-12-19:
Here lies the problem, as already stated by some unknown author:

`issue #484 , quadratic intersection solver is not robust enough for quadratic-quadratic`

Basically, the bi-quadratic simultaneous equations solver (or the quartic equation solver) is faulty.

<hr>

[image]

howtopam, 2021-12-19:
Thank you. I was just testing the 2.2.0-rc3-x86_64.AppImage to see how it was and I encountered a few irregularities that I was able to work arround. The issue of the overlapping elipses not selecting for the divide tool when snap on intersection was selected is resolved by choosing snap on enity. The freezing issue I initally experienced three times has not reappeared after the rebooting of my PC. Everything appears to be O.K. for me to continue using the 2.2.0-rc3-x86_64.AppImage. Again I thank you for time and concern in attempting to locate and resolve these alleged issues.