LibreCAD/LibreCAD #1462: Remote Code Execution vulnerability in LibreCAD 2.2.0-rc3

eldstal, 2021-12-18:
### Vulnerable Products
* LibreCAD 2.2.0-rc3 and older
* Jw_cad 8.24a and older

### Steps to reproduce or sample file

1. Start LibreCAD 2.2.0-rc3 in a debugger
2. File/Open...
3. Unzip and open the attached [proof of concept file](https://github.com/LibreCAD/LibreCAD/files/7739792/jww_cdatamoji.zip)
4. Observe `ACCESS_VIOLATION` crash, with `eip=0x41414141` (AAAA)

Screenshot:

[image]

### Cause
The `CDataMoji` entity deserialization at `LibreCAD/libraries/jwwlib/src/jwwdoc.h` is vulnerable to
a stack buffer overflow.
`char buf[512]` declared in `CDataMoji::Serialize()` on [line 512](https://github.com/LibreCAD/LibreCAD/blob/768285653c46e734a75460928142cda1a33c2c53/libraries/jwwlib/src/jwwdoc.h#L512)
is of fixed size 512. Some varieties of `CDataMoji` provide their own size, e.g. MojiData2 on [line 523](https://github.com/LibreCAD/LibreCAD/blob/768285653c46e734a75460928142cda1a33c2c53/libraries/jwwlib/src/jwwdoc.h#L523)
and no bounds checking is performed. This allows an attacker to overflow `buf` and overwrite other stack variables,
including the return address.

The attached PoC file is tuned to trigger this behavior in the latest windows release of LibreCAD, but the same bug is also present
in older versions and on other platforms.

### Impact
An attacker can craft a JW-CAD input file and thereby gain control over execution flow (EIP controlled directly).

This allows an attacker to run arbitrary code on the system running LibreCAD, with the privileges of the current user.

### Proposed Mitigation
1. Perform bounds checking in `CDataMoji::Serialize()`, and refuse to load the file if it would overflow `buf`.
2. Enable stack smashing protection in the windows build of LibreCAD.

### Operating System and LibreCAD version info
Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

eldstal, 2021-12-18:
A pull request has been submitted with a less invasive mitigation. Data up to 511 bytes is read and any additional data in the field is silently ignored.

Any `jww` file which previously loaded without data corruption should still work with this fix. Files with oversized `CDataMoji` fields will have their data truncated, but should still parse properly.

I've also filed a CVE request with MITRE, referencing this report.

lordofbikes, 2021-12-18:
Just out of curiosity, do you have any JWW files for reference?
I never ever saw one. We even discussed to remove JWW support at all, because the implementer is not responding and nobody has any kind of reference for JWW format.
We also can't say if there were any format changes over time and the LibreCAD implementation still works for latest JWW files.

eldstal, 2021-12-18:
No, I haven't ever used it. [The original application](https://www.jwcad.net/) appears to be alive and well, but I can't say anything about the need for import/export support in FreeCAD.

lordofbikes, 2022-01-04:
fixed with #1463

closed 2022-01-04 by lordofbikes

eldstal, 2022-01-25:
This vulnerability has been assigned [CVE-2021-45341](https://www.cve.org/CVERecord?id=CVE-2021-45341).