LibreCAD/LibreCAD #1464: Remote Code Execution vulnerability in LibreCAD 2.2.0-rc3 (JWW CDataList)

eldstal, 2021-12-18:
# Vulnerable Products
* LibreCAD 2.2.0-rc3 and older

### Steps to reproduce or sample file

1. Start LibreCAD 2.2.0 in a debugger
2. File/Open...
3. Unzip and open the [attached proof-of-concept file](https://github.com/LibreCAD/LibreCAD/files/7740065/jww_cdatalist.zip)

4. Observe `ACCESS_VIOLATION` crash, with `eip=0x41414141` (AAAA)

Screenshot:

[image]

### Cause
The `CDataList` entity deserialization in `LibreCAD/libraries/jwwlib/src/jwwdoc.h` is vulnerable to a stack buffer overflow. `char buf[512]` declared in `CDataList::Serialize()` on [line 784](https://github.com/LibreCAD/LibreCAD/blob/768285653c46e734a75460928142cda1a33c2c53/libraries/jwwlib/src/jwwdoc.h#L784) is of fixed size 512. One variety of `CDataList` provides its own size field, as seen on  [line 795](https://github.com/LibreCAD/LibreCAD/blob/768285653c46e734a75460928142cda1a33c2c53/libraries/jwwlib/src/jwwdoc.h#L795) and no bounds checking is performed. This allows an attacker to overflow `buf` and overwrite other stack variables, including the return address.

The attached PoC file is tuned to trigger this behavior in the latest windows release of LibreCAD, but the same bug is also present
in older versions and on other platforms.

**Note**: This is similar to, but distinct from issue #1462

### Impact
An attacker can craft a JW-CAD input file and thereby gain control over execution flow (EIP controlled directly).

This allows an attacker to run arbitrary code on the system running LibreCAD, with the privileges of the current user.

### Proposed Mitigation
1. Perform bounds checking in `CDataList::Serialize()`, and refuse to load more data to `buf` than actually supported.
2. Enable stack smashing protection in the windows build of LibreCAD.

### Operating System and LibreCAD version info

Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

lordofbikes, 2022-01-04:
fixed with #1465

closed 2022-01-04 by lordofbikes

eldstal, 2022-01-25:
This vulnerability has been assigned [CVE-2021-45342](https://www.cve.org/CVERecord?id=CVE-2021-45342).