LibreCAD/LibreCAD #1468: NULL pointer dereference in DXF parser, HATCH code 93

eldstal, 2021-12-19:
### Steps to reproduce or sample file
1. Unzip and load the [attached proof of concept file](https://github.com/LibreCAD/LibreCAD/files/7741210/dxf_hatch93.zip) in LibreCAD 2.2.0-rc3

### Cause
The `std::shared_ptr` `DRW_Hatch::loop` is written to when loading a `HATCH` entity with code 93. If this occurs before a code 92, the pointer is still `NULL`, leading to a crash.

### Impact
Denial of service.

### Proposed Mitigation
Ensure that `DRW_Hatch::loop` is not NULL before dereferencing at `drw_entities.cpp:1808`

### Operating System and LibreCAD version info
Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

lordofbikes, 2022-01-04:
fixed with #1469

closed 2022-01-04 by lordofbikes

eldstal, 2022-01-25:
This vulnerability has been assigned [CVE-2021-45343](https://www.cve.org/CVERecord?id=CVE-2021-45343).