LibreCAD/LibreCAD #1471: Add the feature of having multiple sheets in one LibreCAD file.

ajk2196, 2021-12-19:
I know that this is my third issue report in one day, but this one too is as important as the previous two. This feature can be seen in many popular CAD softwares like AutoCAD. This feature helps in keeping the drawings well organized within one file itself. As of now, we have to create multiple drawings and keep them sorted in folders.

gerardbm, 2021-12-19:
+1 for this. Every sheet should be able to contain a group of layers (enable/disable layers on every sheet).

melwyncarlo, 2021-12-29:
Perhaps in V3. This is quite substantial.

jbergengh, 2021-12-29:
Alternatively, what if we could do a File > Open, and then select more than one file to open?
Or, taking this one step further, invent a new file type that was a list of multiple .dxf files to open?
This may not get us all the features the OP was looking for, but might be useful and easier to implement.

Comments?

fa201, 2021-12-30:
Please describe a bit more the intended behavior for non-AutoCAD users.

melwyncarlo, 2021-12-30:
This video simply demonstrates what I was suggesting earlier with regards to _tabs_ in FreeCAD. 
Here, each tab deals with an individual file (or in our case, _sheet_).

<hr>

https://user-images.githubusercontent.com/66683108/147727707-f08ac261-1212-41e4-9619-5ea42c023fc9.mp4

melwyncarlo, 2021-12-30:
> Alternatively, what if we could do a File > Open, and then select more than one file to open? Or, taking this one step further, invent a new file type that was a list of multiple .dxf files to open? [...]

The new file type idea is interesting. It wouldn't be costly to implement either. 
It's format would have to be discussed and clearly defined first.

I would suggest the new file format's contents to be : 
* header
* some common properties
* each of the DXF file contents (one after the other)
* footer

gerardbm, 2021-12-30:
The idea of multiple sheets is to separate the model and the presentation. Let me show you a usecase.

We are drawing a 2D home, so we can put the walls in a layer, with a color, line width 0.30, etc. In another layer we can have the dimensions, in blue color, line width 0.10. In another layer we can have furniture, in gray color, line width 0.15. In another layer we have electrical installations, in green color, line width 0.20. In another layer we have water installations, with cold water in blue color, hot water in red color and width 0.20.

All this description is in one tab. The tab with the model, in a black background with multiple layers.

Now we could have another tabs with the presentation. This would be a white background, with the model inserted, so you can select the layers you want to show. For example: in the sheet 1, I want the walls and the dimensions. In the sheet 2 I want the walls with furniture. In the sheet 3 I want the walls layer with the water installations.

So, basically, you will have one model tab and multiple printable tabs. Look how in FreeCAD, he adds multiple template pages to represent the model. https://www.youtube.com/watch?v=TnYMfpdNQoo

The current print preview functionality is not enough for working with LibreCAD. It's very limitative in the workflow.

ajk2196, 2021-12-31:
@gerardbm and @melwyncarlo have got it correct.

One DXF file will contain multiple sheets; and each sheet may have different layers which may or may not be there in another sheets. That is, sheet number 2 may have different layers from the sheet number 1; either none of the layers defined in the previous sheet or some (or all) of the layers defined in the previous sheet.

The sheets should be editable of course. The changes made in any specific sheet must NOT get reflected in the remaining sheets. There should be a feature to rename the sheets. And please, there must be no necessity like the first sheet will have an unchangeable name (Like the layers have: first layer is always '0').

This feature is of great help; will enable storing the data of one entire project in one file itself; and will require opening only once. When there are different files, one has to fish to find the correct file. In real world, projects have a lot of revisions. Sending new file**S** everytime whenever there is a slightest revision is troublesome. This will create a clutter. Sending only one file(no 's' here) with sheets will create less clutter.

melwyncarlo, 2021-12-31:
Your idea is basically a combination of multiple tabs and @jbergengh's combined file format idea.

Furthermore, it's not just tabs, but also something similar to LibreCAD's _layerlist_.
In FreeCAD, this is illustrated in the `Combo View → Model` listbox, alongside tabs.
If what I said is correct, then, basically, we've already got a reference for implementation: FreeCAD.
Though it's debatable whether LibreCAD would want to take that route.

ajk2196, 2021-12-31:
This issue:

https://forum.librecad.org/Import-multi-sheet-dxf-file-td5717806.html#:~:text=LibreCAD%20doesn%27t%20have%20multi,new%20name%2C%20for%20each%20sheet.

I will post more after work.

melwyncarlo, 2022-01-01:
Then, this is something for the `libdxfrw` developers to consider.

ajk2196, 2022-01-02:
Maybe. What I meant by sharing the above thread was that the issue in the thread is something that would get tackled away by what we are wanting to implement. I don't know what 'libdxfrw' is; but if the developers that develop that thing are already busy, we should try to implement it our way; and if we fail; then hand it over to them. This will (sort-of) give them a blueprint of the work.

Or maybe tag a few of those developers in this thread so that this comes to their attention.

So far, the ideas that we were discussing all along the way till the below mentioned @melwyncarlo 's comments were all right and on track.

> Your idea is basically a combination of multiple tabs and @jbergengh's combined file format idea.
> 
> Furthermore, it's not just tabs, but also something similar to LibreCAD's _layerlist_. In FreeCAD, this is illustrated in the `Combo View → Model` listbox, alongside tabs. If what I said is correct, then, basically, we've already got a reference for implementation: FreeCAD. Though it's debatable whether LibreCAD would want to take the _tabs plus list_ route.

melwyncarlo, 2022-01-02:
@ajk2196 `libdxfrw` is a library that is used by the LibreCAD software, but it is developed separately. There is a separate repository for it: [https://github.com/LibreCAD/libdxfrw](https://github.com/LibreCAD/libdxfrw).

@lordofbikes would know more about this.

Firstly, I don't know if multiple sheets is part for the DXF format (it's the first I'm hearing of it). 
If it is part of the DXF standards, then it _might_ already be implemented in `libdxfrw`. 
If it isn't, then this undertaking involves studying the DXF format, defining format standards, 
implementing it the correct way, and then testing it. Not something that can be done right away.
Creating a new DXF standard is like implementing and passing a bill in parliament to become a law.

jbergengh, 2022-01-02:
I still think there is a much easier way to implement this idea.  I think my earlier post was misunderstood so I'd like to explain it in more detail.  Please bear with me.

LibreCAD will allow you to open the same drawing more than once.

Try this:

File > Open > MyDrawing.dxf
File > Open > MyDrawing.dxf
File > Open > MyDrawing.dxf

Drawings > Arrange > Tile Vertically

You'll see three copies of your drawing arranged side by side.
You can now proceed to edit them.  You can make different changes to each of the copies.  For example, per @gerardbm  example, you could edit one of the sheets to show walls and furniture, another to show walls and plumbing, another for walls and electrical, and so on.

Now, save all your changes

File > Save All

LibreCAD will ask you for file names and paths for the modified copies of your drawings.  They will be saved as separate files.

Now, what we need is a way of grouping all the drawings together, so we can load and save them as a group, or project.

I proposed adding a new file type for this.  Let's call it a "Project File".

This Project File is nothing more than a list of the drawings.  It can be a text file, with the contents like this:

MyDrawing.project

/home/me/Documents/LC/MyDrawing1.dxf
/home/me/Documents/LC/MyDrawing2.dxf
/home/me/Documents/LC/examples/myexamples.dxf
/home/me/anywhere/anotherdrawing.dxf

Now, the way this would work is you open LibreCAD, go File > Open > MyDrawing.project, and LibreCAD opens all of the drawings in the project file.

To create a new or save an existing project, you would go File > Save As > Project File.  LibreCAD would save all the drawings currently in session, and also save a Project File containing the list of drawings for that Project.

This, I think, would be very easy to implement.  The question is:  Is it a viable solution for this request?

ajk2196, 2022-01-08:
@melwyncarlo , your suggestion seemed like the right one when you mentioned the reference of FreeCAD. We should go for it before it is too late. @jbergengh , having more than one window open on a medium sized monitor, say 24" won't be good to work on with two or more windows of one document open in a split-screen-like mode. The windows should be minimizable and independent/dependent(as per user's requirements) of/on each other. 

Here is a link to my [Google Drive](https://drive.google.com/file/d/1xbBNx7El3upc_KhZmKcHka2_Co42W-mQ/view?usp=sharing). You will be able to see how it is meant to work; especially in mobile apps made to read dwg and dxf files.