LibreCAD/LibreCAD #1476: Need for Non-scaling dashes' and dots' lines. (Or Proportionately Scaling Dash Lines and Dot Lines)

ajk2196, 2021-12-26:
In current versions of Freecad, when we choose to make or draw dashed/dotted lines, the dashes and dots are drawn, but they appear to be screenspace scaled (the scale of dashes/dots and space between them remains constant irrespective of actual zoom of the entire drawing), which means that the distances between dots and dashes remains same when scaling the drawing in or out. This visual creates a confusion when small areas represented by dashes are placed on extremely large sized plans. I had an experience where I was working on a project with an area of 122m X 249m. It contained some detached buildings. I had set the arc of door openings to 'dashes'. When I zoomed out to see the entire drawing, the dashes appeared as a continuous arc curve. I panicked, but it did go away when I zoomed and found out about this thing. However, wanting to have perfection in work just like all other architects, I had to check all the doors in the drawing which took up considerable amount of time.

Therefore, kindly please set the dash lines and dot lines to scale proportionately with the drawings and remove the feature of screenspace scaling of dashes and dots. Or atleast give the option to the users to whether have screenspace scaling or proportionate scaling.

Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

dellus, 2021-12-30:
Linetype is fixed for printout like linewidth, also regardless of scale. It is not reliable in Print Preview either. 
You have to use your imagination to estimate the appearance in the resulting plan print.
 This may help: https://forum.librecad.org/Dash-dot-line-appear-continuous-tp5718594p5718595.html
Look for the link for linetypes.pdf. I could not upload it here.
It is very unlikely your request will be fulfilled for 2.2.0, as the coding effort would go too deep. Maybe V3.

ajk2196, 2021-12-31:
Having it implemented in V3 would be okay; but if possible, kindly please do it as soon as you good people can. I, posting these many issues feel a bit of guilt: sitting here and just complaining; wish I knew how to code.

Coming back to the main topic; I think that it has to be done some day. Till then, could we get options for tini-er dashes? Because like I mentioned, the dashes in large drawings appear large enough that only I know that these are the dashes. My fellow colleague was confused about why I gave a linebreak in each projection over the windows. I didn't send the file over to the clients and did it in AutoCAD. Thankfully a large portion of the DXF file was read by it and I had to do only a few tweaks in it.

I may not know coding, but I do know a bit of logic of how these things work. If you are familiar with Blender: the 3D software, you would have known the array modifier. It has a setting of creating array of the selected objects till the length of a curve (Certain distance). Similarly, we can code something like creating a basic example of the dashed line and then repeating it over along the length of a line with different length. I see this largely possible and could be quicker to code.

Please look into this. Thank You.