LibreCAD/LibreCAD #1477: Add More Background Grids.

ajk2196, 2021-12-26:
I got this feedback from one of my colleagues; and I felt it too in the beginning that the background grid is not that impressive. Please add the one like AutoCAD.

Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

ajk2196, 2021-12-27:
@fa201 I have updated the contents of my issue. Kindly please remove the 'incomplete' tag now.

fa201, 2021-12-27:
@ajk2196, so your request is to add lines connecting current dots to shape the grid?
Since AutoCAD is your reference, how the user makes the difference between a line of the grid and a line of the drawing?

ajk2196, 2021-12-28:
The lines of the grid will be underlaying the lines of the drawing, always. Moreover, there should be a provision of settings to modify the colour of the grid, like we already have in the current dotted grid. I have already suggested that user should be given the option to choose a certain grid, we don't need to force this grid onto users and force them to use them. If someone finds it interfering, he/she can simply turn off the grid.

Reading your question, I just got an idea that we should add one more option. It should be regarding the glow effect on lines of a drawing whenever mouse is hovered over that part of drawing. This way, it will be easy to tell which line is of drawing. I will raise a separate issue regarding this.

app4soft, 2024-09-01:
### AutoCAD-like Grid Style in LibreCAD

[image]

In "Application Preferences > Appearance" set:

- Background: `#212830`
- Grid Color: `#272d38`
- Metagrid Color: `#323848`
- Highlighted Color: `#ffffff`
- Preview Selectable Refernce Entity Color: `#ffffff`
- Preview Refernce Points: `+` (cross)
- Point Size: `8` Screen %
- [uncheck] Snap indicator line
- [uncheck] Snap indicator shape
- 
In "Current Drawing Preferences" set:

- Grid type: Lines
- Points: `□` (square)
- Point Size: `3` Screen %

(Don't know how to set cursor shape as hollow square, as Preview Refernce Points cross crossing it — is such feature/cursor merged/implemented in latest LibreCAD `2.2.2_alpha`?)

sand1024, 2026-09-26:
lines grid is supported too

closed 2026-09-26 by sand1024