LibreCAD/LibreCAD #1478: Need Better Fonts, A Proper Tool to Convert Fonts We Wish To Use and Improve Anti-Aliasing.

ajk2196, 2021-12-26:
All the fonts that we use in LibreCAD are all outdated-looking. Moreover, they are not even properly anti-aliased. I have gone to the settings and turned on anti-aliasing and restarted the project after saving the settings, but that still didn't give me properly anti-aliased fonts like the ones in AutoCAD.

I tried the 'ttf-to-lff' tool that has been linked-to in some Forums (most probably by Rallaz). Nothing opens up on Windows10. 

Moreover, I have a suggestion. There is an entire library of freefonts (Google Fonts). We should just convert each and every font on this site and host it separately on SourceForge or something like that in the .lff format. **_Doing this will no longer require us to maintain the 'ttf to lff' tool for each and every OS._**

Also, the anti-aliasing feature in LibreCAD is very poor, curves in the drawings that are actually same along their symmetrical axes appear different sometimes on different zoom levels. They do solve up when we zoom in closer, but at first it doesn't, which not only gives a poor impression, but also creates a confusion in the first place.

Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

ajk2196, 2022-01-11:
Please someone pay attention to this. Or atleast give me the tool that does the conversion. I am working on Windows 10. I will convert all good fonts of Google Fonts website and put it up for everyone to use. I tried downloading the 'ttf to lff' tool from the available links; but none of them start. I am always on my desktop wallpaper after double clicking to open it.

jbergengh, 2022-01-11:
@ajk2196 The ttf2lff tool works.  I have converted several TrueType fonts to lff and used them sucessfully in LibreCAD.  You have to run ttf2lff.exe  from a command prompt in Windows.  Click Start and type cmd and press enter.

lordofbikes, 2022-01-11:
@ajk2196 `ttf2lff` is a command line tool which has to be run in a `cmd` window with appropriate parameters.
But before you do any bulk conversion, you should check TTF licenses and conditions. This may also vary from file to file. Even when TTF fonts are free available at Google fonts, doesn't necessarily mean, that they can be shared in converted format. Possibly the fonts have to be packed with author and license information for legal security.

ajk2196, 2022-01-12:
Ok. Thank You @jbergengh  and @lordofbikes . Thats what I was doing wrong; double clicking it to open it; LOL.

closed 2024-05-18 by dxli