LibreCAD/LibreCAD #1480: Add glowing effect to lines / parts of drawings when mouse is hovered upon them.

ajk2196, 2021-12-28:
We should have an option to enable/disable glow effect on lines or parts of drawings whenever mouse is hovered upon them. We should also be able to adjust the intensity of the glow. This will be somewhat helpful if we implement the line grids in next versions of LibreCAD.

ajk2196, 2021-12-31:
Thanks a lot for going through my suggestion and  for putting in the effort to make it possible. Went through your Video. But the part is highlighted (glowed) before the mouse hover and gets un-highlighted (or dimmed/un-glowed) upon hover.

melwyncarlo, 2022-01-07:
@ajk2196  No problem.

The part _is_ highlighted upon hover (and vice-versa); it's just that the highlighted color selected was dull. Bad colour contrast combination chosen for demonstration. If you choose the appropriate _line colour_ and _highlight colour_ in the `Application Preferences`, it should work as desired.

Also, adding the glowing effect is something different. I'll try to figure it out; but this I know, such a feature has to be implemented in the `Paint` library or class.

melwyncarlo, 2022-01-07:
https://user-images.githubusercontent.com/66683108/148492293-1f281aab-39c7-4eb1-bd57-6e72b190481d.mp4

ajk2196, 2022-01-07:
Ok. Thank you. 🍻

melwyncarlo, 2022-01-07:
You're welcome! :smiley:

jbergengh, 2022-01-08:
This one segfaults and crashes after I compile and run it.

melwyncarlo, 2022-01-08:
@jbergengh  Fixed it! Seems like I missed to copy-paste a small code phrase.

I've really got to start testing my solutions in a separate master branch. 
I create my solutions in my main LibreCAD folder where previous solutions have accrued. 
So, it makes it difficult to detect certain errors like this.

jbergengh, 2022-01-08:
Thanks @melwyncarlo I was able to compile and test this now.  I think the feature would be much more useful if the glow effect also happened when selecting entities while using a tool.  For example, if I was using the trim tool to trim two lines together, it would be very very helpful, when positioning the pointer, to see which line will be selected to trim.

melwyncarlo, 2022-01-11:
@jbergengh  Did it! Though it doesn't support `RS_Dimension` and `RS_Insert`, for now. Quite tricky, those ones.

melwyncarlo, 2022-01-11:
Solved them as well!

pollywinters, 2022-04-24:
@melwyncarlo I like the glow effect, but in my build the width of the glowing line appears to be in absolute units, rather than a certain number of screen pixels. That is, as I zoom the view onto a line, the glow gets wider and wider and wider, until the entire screen is wiped out by the glow.
Attached pics show a line segment at initial scale as drawn, plus highlighted version. Then zoomed a bit, you can see highlight is wider. Then zoomed a lot, you can see highlight wipes out so much of screen that it renders the feature useless.
Note: These examples use a line with thickness 0.25 mm. If line width is set to zero then the effect is not so drastic, it seems then that highlight widens as the line is zoomed in on, but only gets to a maximum maybe 20 pixels width (final image).
Also at high zoom factors the screen seems to freeze for 10, 20, 30 seconds when the highligh occurs - possibly the CPU doing some vast amount of off-screen pixel fills.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sand1024, 2026-09-26:
hover highlight is supported, issue is too old

closed 2026-09-26 by sand1024