LibreCAD/LibreCAD #1482: Unit Conversion and Implementation: The most serious issue/bug that LibreCAD should never have

ajk2196, 2021-12-29:
First of all, we need to remove the setting to set units of drawings from two options menu and keep it in any one of them. As of now, we can set the unit of drawings (say foot, meter, millimeter) from both Application Preferences and Current Drawing Preferences. A new comer to LibreCAD will only set it from one of them, because that is the case in every application: You don't have to save a same setting at more than one place; once its done, it takes the effect.

**Secondly, we need to make it work**. This is the only setting that is found more than twice in the combined settings options of LibreCAD, yet doesn't work. When I set the units to meters and draw a square of 2x2 units, it gives me the same-dimensioned square that I draw after selecting 'foot' as the default drawing unit from both menus(Application Preferences and Current Drawing Preferences); that is, a 2 sq.ft. square and a 2 sq. mtr. square overlap each other; while in real world case, the 2 sq. ft. square should be smaller than a 2 sq. mtr. square.

**Steps to reproduce:**

Set default drawing unit to meters in both Application Preferences and Current Drawing Preferences.
Draw a 2x2 square using either line tool or rectangle tool.
Now set default drawing unit to foot in both Application Preferences and Current Drawing Preferences.
Again draw a 2x2 square using either line tool or rectangle tool.
You will see both the squares overlapping.

dellus, 2021-12-29:
The units setting in Application Preferences is for the default setting for new drawings. So if you live in the U.S. you might prefer inches to be the default unit, elsewhere in the world maybe mm. You will get this then with every new drawing out of the box.
The units setting in Current Drawing Preferences, in case you you want to diverge from the default, is saved in that drawing only.

Your second issue: 
Internally LibreCAD thinks in uniform units of same size only. Whatever you have chosen, inches, feet, mm, meters...is more just a naming. That's why the 2x2 meter square and after switching the 2x2 feet square  are the same. There is no automatic conversion. It's the same behaviour in other AutoCAD-like apps.

ajk2196, 2021-12-31:
Yeah, Ok, and I think that implementing this feature of unit conversion will be great for people working to and fro between multiple units. In my country, we follow metric system (mm) in RCC works and the rest like tiles-flooring, masonry and other finishing works are done using units from imperial system (inches and feet). A conversion system that would enable something like allowing drawing a slab in millimeters and then doing the masonry and tiles-flooring work in feet and inches would be of a great help. **As of now, when I try to set the tiles' layout that I have calculated in millimeters because I cannot get thtat done in inches, I always get a small gap beside the last row of the tiles and the wall (the wall too has been done by converting the inches to mm).** Now because of this, I don't know what is right and what is wrong: The tiles' dimensions or the walls'. If I do some adjustments, it will be caught on the site by client's engineers and it would spoil my image as an architect. If I don't do, then the gap would stay there; which too would spoil my image.

**The reason why I am requesting this is that the computers are better at remembering and handling recurring units upto a good place after the decimals; while it is hard for us; probably everyone.** If it was just one case of conversion (say from inches to mm), then I would have never requested for this feature and would have remembered the conversion myself. But this isn't the case. There are a lot of conversions (mm to inches, mm to feet, mm to yards, all of these to and fro with mm replaced by cm and m).

Therefore, kindly please look into implementing this feature. It would be one hell of an achievement to be the first CAD software to have such a feature.

P.S. Also, if there is an architect on Github, he/she will definitely confirm to my problems mentioned above working with multiple units. As I have said earlier, this is serious.

ajk2196, 2022-01-08:
@melwyncarlo can you please see this request?

jbergengh, 2022-01-08:
I've been testing Melwyn's implementation of this request for a while now and it has been working great.  Using it, if you want a 2 meter square for example, you would pick a point and then enter @2m,2m.  If you want a 2 foot square, you enter @2',2'.   You can mix them too.  @2m,2' would get you a rectangle 2 meters wide by 2 feet high.  You can use any units you like.

Is this what you were looking for, or is there a misunderstanding of the request?

There's no telling when, or even if, this change will get accepted.  To get it now you will have to compile the source code.

ajk2196, 2022-01-09:
Well, @melwyncarlo has already solved what you are discussing, and I am very grateful that he did it in no time. Here is its thread: 

https://github.com/LibreCAD/LibreCAD/issues/1483

In this thread, what I mean is, in short, that we should be able to draw in different units (@melwyncarlo solved this) and the outcome zhould be as expected, a square of 1 sq. Ft. should be smaller than a square of 1 sq. Mtr. when drawn in one single drawing without having to close it or change the units settings from the settings. There ahould be one default unit that we can draw in, without having to mention any unit in the command line, but if we want it specifically, we can draw something using inches from command line and give EXPECTED results.

If we draw both the above mentioned squares now in LibreCAD, they would give us same sized squares.

dellus, 2022-01-09:
@jbergengh, as you have been able to test it, what happens when you put dimensions in such a mixed drawing. I assume they all will be in the default unit. I see a following feature request arising....

ajk2196, 2022-01-09:
@dellus @melwyncarlo @jbergengh , here is a simple blueprint of the code we can implement:

1. upon initialization: read the default-set unit and keep using it for all cases when no unit is mentioned in the command line.
2. also upon initialization: declare beforehand the other conversions of the default units; for example:
     if default unit is metres:
     variable named 'inch_to_metre' be assigned a value of 0.0254 since 1 inch is equal to 0.0254 metres.
     another variable named 'feet_to _metre' be assigned a value of 0.3048 since 1 foot is equal to 0.3048 metres.
     another variable named 'centimetre_to_metre' be assigned a value of 0.01 since 1 cm is equal to 0.01 metres.
    similarly, all such conversions' variables should be declared beforehand upon initialization. We should also set a variable called 'metre to metre' and set it to '1'.

Now, there will be two cases:

A: The user won't mention any unit in the command line.
B: The user mentions the unit name in the command line.

In both the cases, the code execution should be set in the following way to give the output expected:

--- BLUEPRINT OF CODE STARTS ---

inch_to_metre = 0.0254
feet_to _metre = 0.3048
centimetre_to_metre = metres

if default unit is metre:
          if ('ft' detected in command line):
          length_of_drawing_line = (input_in_command_line) * feet_to_metre

          elif ('cm' detected in command line):
          length_of_drawing_line = (input_in_command_line) * centimetre_to_metre
          .
          .
          .
          .
          .
          else:
          length_of_drawing_line = (input_in_command_line) * metre_to_metre

elif default unit is feet:
          [ all similar logic explained above ]

--- BLUEPRINT OF CODE ENDS ---

jbergengh, 2022-01-09:
@dellus yes, the drawing can still only have one default unit.  Dimensions still use the default unit, although you could edit the dimension properties and change the value if you had to use mixed units on the same drawing.  Melwyn's implementation simply does the conversion for you.  If your drawing unit is mm, and you draw a rectangle @1in,1in, it converts that to a rectangle 25.4mm x 25.4mm, and a horiz/vert dimension would be 25.4.

ajk2196, 2022-01-09:
@jbergengh If you are saying that it converts it, then its great but not great enough, as it only does this for inches and feet when the default unit has been set to something apart from inches and feet. In case where majority of the dimensions in a drawing are given in feet and inches and few of them are given in metres, cm or mm, then a likely case would be to set the default unit to either foot and inches and then draw the few drawing parts in mm, cm or metres, the functionality which we are trying to add via this thread.

As I have commented 2 comments earlier a blueprint of the code we can implement to add this feature, we should simple be able to do it.

thecadman2017, 2022-01-09:
Hi I am new to Github but used it for trying to learn how to use the small
pcb and learning the C language. I am trying to find a CNC cad/cam program
that I dont have to pay thousands of $$$
Can you help me in any way.
Thank you
Manny
Don't Ever Give Up !!!

ajk2196, 2022-01-11:
How to delete thecadman2017's comment above so that we don't have spam in this thread? Please read  two of my previous comments; one of them contains the blueprint to code implementation.

beju1711, 2023-03-20:
Hello,
I followed the previous discussion and understood the requests of ajk2196 very well, because for years I have the same issues dealing with the units. I just installed the new LIbreCAD version 2.2.0 hoping that especially the change from mm to inch and vice versa works as expected but in my opinion it doesn't. 
I'll give an example: By default I work with millimeters. In this case everything is ok. The dxf contains all objects measured in millimeters. I am fine with this because when I feed my CAM tool (Estlcam) with the dxf-file the CAM tool sees the elements coming along in millimeters. Then I change the units settings in LC to inch, e. g. because I need to cut some PCBs which are based on 2.54mm grid. Again I feed my CAM tool with the according dxf-file. Surprise, surprise: what the CAM tool sees is again all dimensions in mm. I tried the same with food, meters and centimeters. In all cases the millimeters remain tenaciously. Maybe I do something wrong. Who can help me to leave the trap if there is one?
Thanks and BR, B.