LibreCAD/LibreCAD #1483: Need support for working with feet and inches.

ajk2196, 2021-12-29:
As of now, we cannot work with feet and inches in LibreCAD. Using the co-ordinate system, if I type in ' 2'6" , 8'6" ', it throws expression syntax error. Not supporting this feature is a serious drawback of LibreCAD as even today, many architects work in imperial system either by choice or compulsion. I myself work in imperial systems because I have to; the dimensions of tiles, formworks and all other Construction goods are manufactured in feet and inches.

Therefore, kindly please look into implementing this feature.

Steps to reproduce or sample file:

Explained above.

Operating System and LibreCAD version info

Version: 2.2.0-rc3
Compiler: GNU GCC 7.3.0
Compiled on: Nov 29 2021
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

melwyncarlo, 2021-12-29:
* This issue was already brought up in #1372 and solved in #1413.

I have just implemented your request for solving inches and feet in their combined format (e.g. `1'2"`).

<hr>

https://user-images.githubusercontent.com/66683108/147682594-eb1d1a8e-4e5b-4d52-9c11-7f832b03bd0d.mp4

ajk2196, 2021-12-29:
Thanks a lot Melwyn Carlo. This was just one part of a somewhat greater issue. [This](https://github.com/LibreCAD/LibreCAD/issues/1482) is something that should work along with what you have implemented. I hope that it gets solved and can be used together without any further bugs. Moreover, when will we be able to use this feature of inputting data and getting results in feet and inches?

melwyncarlo, 2021-12-30:
You're welcome, @ajk2196 ! :smiley: 
With regards to its official implementation, I've got no clue, whatsoever.

ajk2196, 2021-12-31:
Ok; thank you.