LibreCAD/LibreCAD #1488: Fonts are no longer working

jirislaby, 2022-01-01:
After commit e666fb172e3497dd00435ff435cc6d5177f3b8ee, fonts are not loaded as the paths in `RS_System::getDirectoryList` are incorrect. Like:
```
dirList ("/home/xslaby/.librecad/fonts", "/home/share/doc/librecad/fonts", "/home/share/librecad/fonts", "/home/lib/librecad/fonts")
```
Before that commit, the code did:
```
dirList.append( "/usr/share/" + appDirName + "/" + subDirectory);
```
which resulted in correct `/usr/share/librecad/fonts`.

lordofbikes, 2022-01-04:
The basic idea behind this change is, that an AppImage is mounted as loop device in `/tmp` in a randomly named folder. Inside this folder we have a root file system.
So the concept is to read the executable path on start up and find system folders relative to the executable.
This should then work for AppImage and distribution packages.

Usually packet maintainers will place `librecad` executable in `/usr/bin`. LibreCAD then search for resources in `/usr/bin/../share/librecad/` which results in `/usr/share/librecad/`.
This is what you expected.
But from your list I assume that the `librecad` executable is in your home folder `/home/xslaby/` which will result in `/home/xslaby/../share/librecad/` which is `/home/share/librecad/` effectively.

So the problem is presumably the correct placement after building from source. When you have fonts and other things in `/usr/share/librecad`, you should place `librecad` in `/usr/bin`, then it should work.

There is the script `scripts/postprocess-unix.sh` which collects everything needed in `[LibreCAD_ROOT/unix]` after build. This folder contains fonts, patterns, libraries and more.
Running `librecad` from this folder works for me well. This is also the described way in build instructions.

So the options I see (for self builder) are to run `librecad` in place with `[LibreCAD-ROOT]/unix/librecad` or move the whole `unix` folder to any other place and run it from there.

We can also add instructions, where to place the needed files and folders. But this may vary on different distributions and also need `sudo` permission, e.g. to copy/link files to `/usr`.

You can take a look at [CI/build-appimg.sh](https://github.com/LibreCAD/LibreCAD/blob/70f7a86ba991b11db05c4f392d516d12fd98b7e8/CI/build-appimg.sh) to see where all the resources from `unix` folder are copied to. Remove `appdir` from  destination paths and you see the `/usr` paths. These paths are usually used on Debian based distributions.

jirislaby, 2022-01-05:
@lordofbikes Sorry for not being specific, I am using lirbecad from openSUSE Tumbleweed. I.e. it *is* installed in `/usr/bin`.

Note that `prgDir` is the current directory, not `/usr/bin`. You'd have to start librecad as `/usr/bin/librecad` to `prgDir` be `/usr/bin`:
https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/main/main.cpp#L209-L212

lordofbikes, 2022-01-05:
Finally I got it, the missing link was:
```sh
$ export PATH=$PATH:/tmp/q
```
`argv[0]` does not contain any path information when the executable is located through `$PATH`.
`QFileInfo::absolutePath()` then seems to fallback to `pwd`, which explains your observations.

I already used `QCoreApplication::applicationDirPath()` in https://github.com/LibreCAD/LibreCAD/commit/e666fb172e3497dd00435ff435cc6d5177f3b8ee, so this should have worked. But I removed it in https://github.com/LibreCAD/LibreCAD/commit/a64f1f481cd3316e9cce79beeeafd59faf67b703 because `appDir` and `binDir` was always the same for me.

There are a couple conditions in `RS_System::getDirectoryList()` I already wondered about, but now they makes sense. Some of them seem obsolete when using `QCoreApplication::applicationDirPath()`, I'll check this and it will need some detailed testing then.

closed 2022-01-06 by lordofbikes

u2fly, 2023-06-07:
@lordofbikes, Latest `2.2.1-alpha` AppImage **got this issue again**:

-  `LibreCAD-2.2.1_alpha-125-ga28f13b5-x86_64.AppImage`

commit 8c3b652a (2022-01-06): fixed #1488, getDirectoryList() failed on Linux when librecad is in $PATH