LibreCAD/LibreCAD #1489: Missing dimension text and text style

rola22, 2022-01-03:
Hello.

Today i downloaded and compiled latest version of LibreCAD.
What i found is that i cannot see dimension text. I opened Options -> Current Drawing Preferences -> Dimensions
and i didn't found Text style for dimension, drop-down list is empty.

I'm using Linux Mint 20.2 Cinnamon with 5.11.0-43-generic  #kernel

[image]

rola22, 2022-01-03:
And looks like i found issue - it was me.
I've compiled Librecad on my secondary drive, which is in home/.../media/...

When i compiled it again on my main drive everything is ok.

But what i want to use my second drive for compiled programs? How to compile them correctly?

fa201, 2022-01-03:
I can compile on secondary drive hosting my home directory on Ubuntu.
Did you try this proceedure: https://librecad.readthedocs.io/en/latest/appx/build.html?highlight=compile#building-on-linux

rola22, 2022-01-03:
Yes, i was following this guide but instead of downloading zip file i cloned source code.

lordofbikes, 2022-01-03:
This is possibly related with #1488

lordofbikes, 2022-01-04:
I checked this, it is indeed a duplicate of #1488, please follow there

closed 2022-01-04 by lordofbikes