LibreCAD/LibreCAD #1493: Implement a robust SVG writer / converter.

melwyncarlo, 2022-01-06:
The current SVG exporter does not export `RS_EntityLine`, though it is claims to currently support it:

<br>

https://github.com/LibreCAD/LibreCAD/blob/2c2c1b0d44fdaf9af20edabdae08a66a05ca228d/librecad/src/lib/generators/lc_makercamsvg.cpp#L250-L292

<br>

Also, it needs to support `RS_EntityDim...`s.

lordofbikes, 2022-01-06:
The intention of SVG export is to export manufacturing data for machining post processors. And for that dimensions are not needed, even disturbing.
On the other hand, most DTP software, with SVG as basic file format, have DXF import filters, which import everything from DXF.
So I wonder if improvement is really needed here or if it is just reinventing the wheel.

melwyncarlo, 2022-01-06:
When I exported in SVG using the export images menu, the generated arc image was _jagged_. 

<hr>

[image]

<hr>

[image]

<hr>

It uses `RS_PainterQt` class, which uses `polyline` to draws _arcs_; it makes no sense!

https://github.com/LibreCAD/LibreCAD/blob/2c2c1b0d44fdaf9af20edabdae08a66a05ca228d/librecad/src/lib/gui/rs_painterqt.cpp#L173

https://github.com/LibreCAD/LibreCAD/blob/2c2c1b0d44fdaf9af20edabdae08a66a05ca228d/librecad/src/lib/gui/rs_painterqt.cpp#L228

<hr>

[image]

[image]

<hr>

The SVG generated in the CAM/SVG menu was superb and clean!
Just a few lines of neat and exact code.

<hr>

[image]

<hr>

Shouldn't the two operations share the same code, atleast with regards to writing SVG code?

dellus, 2022-01-06:
SVG export as it is now is not usable for DTP or advanced graphics anyway. It doesn't show linewidth properly, only one width for all lines. That's why I always export to pdf first, and then open this in Inkscape for further processing and saving as svg.
You have this jagginess in arcs too via pdf, but much finer. By the way, it is visible in text (dimensions) also.
Importing a LibreCAD dxf into Inkscape isn't a solution either, as it will look quite off, for instance text, dimensions, and hatches are not shown properly. 
In svg export you can obtain finer jagginess also, by setting a higher resolution, at the cost of larger file size.

melwyncarlo, 2022-01-07:
Thank you very much for the workaround. But I think this problem might quite easily be solved.

There are two separate SVG compiling functions in LibreCAD: one produces jagginess, and the other does not.

The `CAM/Plain SVG` does not create the jagginess. So, what I am suggesting is that, why can't we divert the export image's SVG code writing function to the SVG writing function used in `CAM/Plain SVG`?

dellus, 2022-01-07:
Well, if you can get it done, super duper!

melwyncarlo, 2022-01-09:
I wish to extract the SVG writing functions within the `CAM Maker SVG` class, and create a new `SVG Graphics Creator` class, and extend those functions to incorporate SVGs for DTP as well.

Should this be a feature request?

dellus, 2022-01-09:
SVGs, DTP....Is there someone thinking about producing a LibreCAD book?