LibreCAD/LibreCAD #1494: Able to draw into a locked layer

gargan26, 2022-01-06:
### Expected behavior
When a layer is locked, I expect not to be able to draw elements into it.

### Observed behavior
When the selected layer is locked, I am able to draw elements into it.

### Steps to reproduce or sample file
1. Lock layer
2. Draw elements with locked layer selected.
3. Elements placed into locked layer regardless of locked status.

### Operating System and LibreCAD version info
Arch Linux

LibreCAD Version Info:
Version: 2.1.3
Compiler: GNU GCC 11.1.0
Compiled on: Jun 10 2021
Qt Version: 5.15.2
Boost Version: 1.76.0
System: Arch Linux

dellus, 2022-01-06:
There is a pull request submitted adressing this issue : #1460.
But it seems it is standard behaviour in other dwg/dxf based cad apps to be able to create entities on locked or unvisible/switched off layers. You just can't select and edit entities on an off layer.
So maybe this pull request will be rejected.
But QCad for instance denies drawing on locked or off layers. So maybe LibreCAD will follow this path.

fa201, 2022-01-07:
Documentation specifies that the change is not permitted, so it is in line with the behavior.
https://librecad.readthedocs.io/en/latest/ref/widgets.html#layer-list-dock

I use a lot the lock/unlock layers and the current behavior is good or bad whether:
- I know what I do and want to create but not modify on the frozen layer.
- I make a mistake and create on the wrong layer. I have to unfreeze to delete and freeze back.

Basically current or improved behaviors will be OK for me.