LibreCAD/LibreCAD #1496: Fillets cause spurious line(s)

JohnCC330, 2022-01-07:
### Expected behavior

Clean drawing with rounded corners

### Observed behavior

New lines appear when changing line style

### Steps to reproduce or sample file

Attached a sequence of operations which cause the effect:
1) Start with a 10x10mm rectangle
2) Add fillets (in my case 4mm)
3) Change properties - line style - to medium dash - new lines appear
4) Change props back to continuous - lines remain (plural - there seem to be a double line)

### Operating System and LibreCAD version info

Linux Slackware64-current
LibreCAD 2.2.0alpha.
[image]

melwyncarlo, 2022-01-07:
@JohnCC330  It seems to work fine in `Edit layer attributes` and in `Toolbar`. Perhaps a video would help?

dellus, 2022-01-07:
I can confirm with LC 2.2.0 rc3. It seems to have to do with the rectangle being a polyline. If it is exploded first it doesn't happen.

jbergengh, 2022-01-07:
https://user-images.githubusercontent.com/55407804/148573384-d094cda5-f0b4-42d3-83e4-73ceae9be44a.mp4

JohnCC330, 2022-01-07:
Sorry, have been out - life goes on... Thanks for the confirmation, and the video - it shows exactly what happens.

Exploding afterwards permits removing the extra lines. I have the impression that this did not happen in
previous versions (say 2.1.xxx)

melwyncarlo, 2022-01-08:
@JohnCC330 Now, it should no longer produce the erroneous line upon changing the line type.

JohnCC330, 2022-01-09:
Cheers! Will recompile ASAP.

commit 12dc3ea1 (2022-06-06): Merge pull request #1503 from melwyncarlo/Task_53

fillets caused spurious lines, fixes #1496
closed 2022-06-06 by lordofbikes