LibreCAD/LibreCAD #1499: Copy-pasting a rotated insert, rotates the duplicate parent block too

melwyncarlo, 2022-01-08:
I believe that the parent block should remain intact, just like its predecessor, 
and that only the insert should rotate by the angle set by _its_ predecessor.

Basically, the pasted block and insert's orientation should be the same as that of their copied counterparts.

<hr>

https://user-images.githubusercontent.com/66683108/148632363-1e90eda0-40a8-4811-baca-c6f753882e94.mp4

melwyncarlo, 2022-01-09:
Is this a known bug, or is it the intended function?

OleksiiSalnikov, 2022-09-29:
Hello! Found this behavior still reproducible in 2.2.0rc4.
 Try avoid by importing saved to file block:
1. Have a block with drawing
2. In a blocks list select that block
3. Use save block button to produce file
4. In a different document use file-import-block and point to file from step 3