LibreCAD/LibreCAD #1504: Mouse wheel zoom doesn't center at mouse position

dellus, 2022-01-08:
### Expected behavior: Zooming in and out using the mousewheel should center at the mouse position.

### Observed behavior: Zooming in centers at mouse position only if mouse is at screen center. The more the mouse is off the screen center the zooming behavior gets more unpredictable. Still worse zooming out does behave in a different way so you never get back to your start view. 

### Operating System and LibreCAD version info: V 2.1.3, V 2.2.0

Also see this discussion: https://forum.librecad.org/Zoom-in-out-with-mouse-wheel-doesn-t-perform-the-same-as-zoom-in-out-with-CTRL-tp5721198.html
In older versions the mouse wheel zoom centers at mouse position and CTRL +/- centers at screen center. Makes sense, doesn't it? Even if the current way is by purpose I propose to return to the previous procedure as I consider it the better solution.

melwyncarlo, 2022-01-09:
This seems like the intended behaviour; the screen zooms towards and into the current mouse position.

<hr>

https://user-images.githubusercontent.com/66683108/148682294-99ffe4eb-936b-4fad-a8b5-44d9780287a8.mp4

dellus, 2022-01-09:
Try it with the object of interest in a corner of the screen at the beginning, zoom in and then out again, you will not get the original view back.
Also compare with how it works with control and +/-, this is how it should be.

closed 2023-06-21 by dxli