LibreCAD/LibreCAD #1507: Remove unnecessary Free Snap button

dellus, 2022-01-12:
### Expected behavior
Free snap is built in all time behavior
### Observed behavior
It has to be set activated
### Operating System and LibreCAD version info
All OSes and versions

Often Newbies are wondering why dimension lines are sitting directly on the dimensioned object, instead of a bit in a distance. It happens when Free Snap is off, so the dimension line sticks to the picked object.
Is there any scenario where it makes sense to have a forced snap?
If not I propose to have free snap fixed and drop the button completely.

fa201, 2022-01-12:
I switch off *free snap* most of the time as the "size" of snap seems to depend on zoom factor. I mean if I want to have *free snap* activated I have to zoom to *intersection*/ *extremities* / *centers* before every click which is painful. I use LibreCAD mainly in mechanical engineering where accuracy does matter.

But we could set free snap on by default for the first install ?

dellus, 2022-01-12:
I have noticed too that the size of the snap area seems to vary with zooming, and it seems to be randomly. Still I have never felt it necessary to switch free snap off. I don't have to zoom in to the desired snap points, just move the cursor to them.
 Free snap doesn't spoil accuracy, I don't understand your argument here. But if there are users who appreciate the possibility to switch it off, then it should stay.
In other CADs you can have a snap area frame, with a size in screen pixels you can set, and is reliable. In my view this would be the best solution, but probably requires too much reconstruction for LC 2.2.0.

jbergengh, 2022-01-14:
I leave free snap on almost all the time, but I always turn it off when I'm using Spline through Points.  It makes picking each point much more quick and precise.

dxli, 2024-05-18:
in 2.2.1 and later, you may choose "Exclusive" for snapping. Then, "Free Snap" is needed.

Without "Exclusive", "Free Snap" is allowed to be used with other snap methods:

For example, if "Free" and "End points" are both enabled, "Free snap" is only used when away from an end point.

closed 2024-05-18 by dxli