LibreCAD/LibreCAD #1510: Spline through points not working in 2.2.0

Kevin1375, 2022-01-13:
Expected behavior:
When I select "Spline through points", I expect a spline to be created after escaping the sequence (as 2.1.3 behaves).

Observed behavior:
No spline is created.

Steps to observe behavior:
Create new file.  Create a number of points.  Select "Spline through points" and click on the points previously created.  A spline appears as points are clicked, but disappears once you escape from the sequence.  (In version 2.1.3, a spline is created.)

Operating System and LibreCAD version info:
Version: 2.2.0-rc3-21-g5d331c8b
Compiler: GNU GCC 7.3.0
Compiled on: Jan 10 2022
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

I could return to 2.1.3, but I had issues with ANSI B custom paper sizes (issue #1176).  Is there a version that allows custom paper sizing and spline through points?

dellus, 2022-01-13:
The procedure has been changed. Escape doesn't work anymore, press right mouse button to finish the operation.
In the status bar the text right to the mouse symbol should read "Finish" instead of "Back". I remember to have seen "Finish" there at some other command, but don't remember which.

lordofbikes, 2022-01-13:
@dellus is right, this was just discussed in #1373, another tool, but same issue.
And there is more explanation in Librecad/docs#101.
For _Spline trough points_ I just checked, the right button info is **back** and probably misleading. Seems **finish** is more intuitive, if anybody recognize it at all. :rofl:
So we keep this open for tracking, to check the right mouse button labels, for other tools too.

jbergengh, 2022-01-13:
Here are all the mislabeled and missing right mouse button info I could find:

Spline through points tool:  **Back** label should be **Finish**
Spline tool: **Back** label should be **Finish**
Polyline tool:  **Back** label should be **Finish**
Polygon (Cen,Cor) tool: Has no labels. Should be **Back** and **Cancel**, same as the Polygon (Cor,Cor) tool
Polygon (Cen,Tan) tool: Has no labels. Should be **Back** and **Cancel**, same as the Polygon (Cor,Cor) tool

Kevin1375, 2022-01-14:
Thanks for the explanation.  I confirm right click works for me.

Ratty1978, 2024-01-27:
Glad I finally found this.  Right Click to "end" a Spline (Windows 11 LibreCad 2.2.0.2).   I'd been just drawing multiple arcs!

closed 2026-09-26 by sand1024