LibreCAD/LibreCAD #1512: Viewing Current Drawing Preferences changes dimension text alignment

jbergengh, 2022-01-14:
Simply looking at Current Drawing Preferences, without changing anything, wipes out all Dimension text alignment changes

### Expected behavior
Existing dimensions on the drawing should not be modified or updated unless dimension properties are actually changed in drawing preferences.

### Observed behavior
Dimensions are updated, wiping out any text alignment changes, even when the user makes no changes to preferences

### Steps to reproduce or sample file
Draw some dimensions.
Move the dimension text
Options > Current Drawing Preferences >  make NO CHANGES > click OK
Observe: Dimension text moves back to default location

See video attachment

### Operating System and LibreCAD version info
Linux Mint 20.3
LC 2.2.0-rc3
GNU GCC 9.3.0
QT 5.12.9
Boost 1.75.0

https://user-images.githubusercontent.com/55407804/149429395-e30e5f6e-6a39-4752-8c1a-43d2b1a160fc.mp4

dellus, 2022-01-14:
You can avoid this by pressing "Cancel" instead of "OK".
I guess with OK everything is reset as if it is a new setting. Bad if you have a lot to repair.

jbergengh, 2022-01-14:
I discovered this when i went in to change my paper size, and my dimensions changed.  Didn't touch anything under the dimensions tab. Had to hit OK to change the paper size.

dellus, 2022-01-15:
I have already tested it on my own. Yes, you are right. Though the title of the issue is not accurate ;-) . You did not only view  Current Drawing Preferences, you indeed did change something: the paper size! And hit OK.
Maybe this is not so easy to fix as it might be rooted in the way moving dimension text is coded. But it really is a  problem.

jbergengh, 2022-01-15:
I've certainly got no problem updating the title if you can think of a better one :-)

But the dimensions should not get modified for no reason.

Options > Current Drawing Preferences > **make no changes whatsoever (do not change paper size)** > click OK

Doing that should not change dimensions, or anything at all for that matter.  If no changes were made to any preferences then clicking OK should be the same as clicking Cancel.  It certainly should not move dimension text.

This should be a simple fix (I think).  If the user changes anything under the Dimension tab, set a flag.  When he clicks OK, if the flag is set, then update the dimensions to the new preferences.  Otherwise don't.